raspberrypi/pico-sdk #215: Recognize "standard" run-from-ram binaries from bootloader?

WestfW, 2021-03-03:
I realize that this would mean a ROM change, and I'm not sure how important it is, but...

As I understand it, if you load a UF2 file that contains only RAM content, the bootloader will assume that it is a run-from-ram binary and start it at the lowest address that has been loaded:
https://github.com/raspberrypi/pico-bootrom/blob/d37cc82d584b51ec86f5fc4959fab660c3044fd1/bootrom/virtual_disk.c#L161

```
       if (!task->result && _uf2_info.valid_block_count == _uf2_info.num_blocks) {
            safe_reboot(_uf2_info.ram ? _uf2_info.lowest_addr : 0, SRAM_END, 1000); //300); // reboot in 300 ms
        }
```

This is ok, but is incompatible with "standard" cortex binaries that have the "initial stack pointer" in the vector table at their lowest location.
I think it would only require a little bit of code for either the WD reset code, or the bootrom that sets it up, to notice that the first lword in the binary looks more like a Initial SP value than code, and take appropriate action to support bother traditional cortex binaries as well as the Pico-SDK style RAM binaries.

lurch, 2021-03-03:
See also #203

kilograham, 2021-03-03:
Yup; i have thought about this too - we obviously wont be doing a new ROM revision just for that though! It's on the list