raspberrypi/pico-sdk #217: hardware/pwm.h: No such file or directory

camielverdult, 2021-03-03:
Hi all,

I am using this CMake file in CLion (with a toolchain + configuration that uses the gcc-arm compiler).

```cmake
cmake_minimum_required(VERSION 3.17)

set(PICO_SDK_PATH ~/pico-sdk)

# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(${PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(pico)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

set(CMAKE_CXX_STANDARD 20)

add_executable(pico main.cpp)

target_link_libraries(pico pico_stdlib)

pico_add_extra_outputs(pico)

enable_language( C CXX ASM )
```

This compiles the code snippet under step 4 of the Quick Start. After copying the code of [this example (pwm)](https://github.com/raspberrypi/pico-examples/blob/master/pwm/led_fade/pwm_led_fade.c). This is the include section from that example code:
```cpp
#include "pico/stdlib.h"
#include <stdio.h>
#include "pico/time.h"
#include "hardware/irq.h"
#include <hardware/pwm.h>
```

When trying to compile the code example, gcc throws an error about hardware/pwm.h not being a file or directory. It does not complain about hardware/irq.h while both headers have the same structure. Is this due to a setup fault? How could I fix this?
[image]

[image]

lurch, 2021-03-03:
Does it work if you change `#include <hardware/pwm.h>` to `#include "hardware/pwm.h"` ?

camielverdult, 2021-03-03:
> Does it work if you change `#include <hardware/pwm.h>` to `#include "hardware/pwm.h"` ?

It does not no, I have tried a bunch of different includes. Seems like my CMakeList just does not include all of hardware parts. Trying to compile via chapter 9 in the documentation seems to work fine using VSCode.

lurch, 2021-03-03:
Can you copy'n'paste the compiler error you get after changing the `<` and `>` to regular double-quotes please?

camielverdult, 2021-03-03:
> Can you copy'n'paste the compiler error you get after changing the `<` and `>` to regular double-quotes please?

```
====================[ Build | pico | ARM ]======================================
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/camielverdult/Desktop/pico/cmake-build-arm --target pico -- -j 6
[  1%] Performing build step for 'ELF2UF2Build'
[  5%] Built target bs2_default
[  8%] Built target bs2_default_padded_checksummed_asm
[100%] Built target elf2uf2
[ 10%] No install step for 'ELF2UF2Build'
[ 11%] Completed 'ELF2UF2Build'
[ 20%] Built target ELF2UF2Build
Scanning dependencies of target pico
[ 22%] Building CXX object CMakeFiles/pico.dir/main.cpp.obj
/Users/camielverdult/Desktop/pico/main.cpp:14:10: fatal error: hardware/pwm.h: No such file or directory
   14 | #include "hardware/pwm.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
gmake[3]: *** [CMakeFiles/pico.dir/build.make:83: CMakeFiles/pico.dir/main.cpp.obj] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:1495: CMakeFiles/pico.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:1502: CMakeFiles/pico.dir/rule] Error 2
gmake: *** [Makefile:151: pico] Error 2

```

lurch, 2021-03-03:
Oh, you'll also need to add `hardware_pwm` to your `target_link_libraries` in `CMakeLists.txt` - see  https://github.com/raspberrypi/pico-examples/blob/master/pwm/led_fade/CMakeLists.txt

camielverdult, 2021-03-03:
Changing
```cmake
target_link_libraries(pico pico_stdlib)
```
to
```
target_link_libraries(pico pico_stdlib hardware_pwm)
```
solves the error, and it compiles. Thank you!

closed 2021-03-03 by aallan