raspberrypi/pico-sdk #224: Suggestion: Add a putchar_raw() or similar function for sending bytes over Serial without CR/LF

ZodiusInfuser, 2021-03-04:
I was writing a serial passthrough program on Pico for programming another micro and experienced issues with the returned data containing `0x0d` before `0x0a` when appearing on my PCs terminal. After spending over 3 hours discounting the usual physical wiring and baudrate issues, I concluded the issue must be with how `putchar()` sends the data from Pico.

After searching for an equivalent to `uart_putc_raw()` I ended up resorting to the Pico community on Discord, who informed me that CR/LF conversion is on by default so I had to call `stdio_set_translate_crlf(&stdio_usb, false);` to disable it.

It was in no way obvious that this was the issue, nor that the above was the function to call to address it.

I have got my program working now, but to make people's lives easier in the future, I'd like to suggest including a `putchar_raw()` function that ignores the translation for that specific byte, as that would then be consistent with the uart portion of the SDK.

Also as a general comment (maybe worth a separate issue in pico_feedback?), some documentation on functions that Pico exposes from the standard library would be nice (e.g. in Chapter 2.7.1 of C++ SDK). Someone coming from Arduino would not know these, nor know to search for them. For instance, `putchar()` has no mention at all, so I only found it by trying the `puts` that the SDK does mention.

Thanks

closed 2021-10-12 by kilograham

FilipDominec, 2022-01-18:
Sorry for my late commenting. 

While this is a very good idea, ```putchar()``` operates bytewise and appears much slower than ```fwrite(output_buffer, length, 1, stdout)```, the latter thus being preferable full-speed USB transfers (sustained 850 kBps). But ```fwrite``` is also affected by CR/LF mangling unless this is disabled explicitly. 

Is there any equivalent like ```fwrite_raw()``` planned?

kilograham, 2022-01-18:
if you don't need CR/LF i would just use stdio_set_translate_crlf(&stdio_usb, false);