raspberrypi/pico-sdk #235: multicore_fifo_wready() docs might be backwards?

davidgiven, 2021-03-05:
Filing as a bug rather than just sending a pull request because this might be intentional, but the documentation for `multicore_fifo_wready()` says:

```
Check the FIFO to see if the write FIFO is full.
Returns true if the FIFO is full, false otherwise
```

AFAICT it returns true if the FIFO is _not full_ (i.e. has at least one slot ready to write), which is what the name suggests. If this is correct, the documentation needs to be corrected; if not, the name probably wants changing!

lurch, 2021-03-08:
Thanks, the [RP2040 datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) says:
```
1    RDY     Value is 1 if this core’s TX FIFO is not full (i.e. if FIFO_WR is ready for more data)   RO    0x1
```
so it looks like this was a simple typo. Fixed by #239

Wren6991, 2021-03-10:
Thanks, fixed

PR #239 (2021-03-10): Fixup incorrect doxygen for multicore_fifo_wready
Fixes #235
closed 2021-03-10 by Wren6991