raspberrypi/pico-sdk #238: I2C support for short writes

dhalbert, 2021-03-07:
I was told by word of mouth that the `DW_apb_i2c` hardware design licensed for the RP2040 does not work for I2C write transactions for 0 to 2 bytes of data (not including the address).  MicroPython special-cases this by [doing bitbang I2C for writes of length 2 or less](https://github.com/micropython/micropython/blob/680ce45323248df288ee8ebd055a4caacb3e46f3/ports/rp2/machine_i2c.c#L124-L144), and we adopted [the same thing](https://github.com/adafruit/circuitpython/blob/a4133c4545ada7c9ef0d65c675fcbf24e9d15f49/ports/raspberrypi/common-hal/busio/I2C.c#L162-L180) in CircuitPython. It took a little painful experimentation to get it exactly right.

`pico-sdk` does not do anything special to handle short writes. Should it? Or should it be documented that they don't work, in the datasheet and/or the SDK doc? Or is this actually not a hardware issue, but a hardware-use that could be worked around in another way? I don't have access to the support info from Synopsys for `DW_apb_i2c` to clarify this. I know this is a commonly licensed piece of IP, but when doing webesarches I haven't found specific reference to this issue elsewhere. Thanks for any further info you may have on this.

aallan, 2021-03-08:
We've discussed this internally and we're not aware of the issue. Can you provide a case that triggers the need to special case writes of 2 bytes or less?

dhalbert, 2021-03-08:
(I was working up a simple example for 0-byte writes but then you edited your response. :slightly_smiling_face:  )

> We can't do 0-byte writes (this is a known and intentional limitation, not a bug)

I poked around some more and found this comment in `i2c.c`, which I had missed before:
```c
    // Synopsys hw accepts start/stop flags alongside data items in the same
    // FIFO word, so no 0 byte transfers.
```
So that explains it. I think it would be good to have this noted in the SDK docs and the datasheet. I can open an issue in `pico-feedback` and/or here for that, if you'd like.

As for 1 and 2-byte writes, I apologize for the false alarm. @dpgeorge has chosen to do 0-2 byte writes as bitbang in MicroPython. I tried to vet his choice and thought I saw that 1 and 2 byte writes were not working, but I just retested and I don't see the problem. (This was in parallel with [trying to get the TCS34725 sensor](https://github.com/adafruit/circuitpython/issues/4082) working on RP2040 I2C: it does not, for reasons I have not yet discerned, though it's fine on other ports. It does NACKS in cases where the data sent looks fine.). Thanks again.

aallan, 2021-03-08:
>  I think it would be good to have this noted in the SDK docs and the datasheet. I can open an issue in pico-feedback…

Thanks, if you could do that, that'd be great. I'll close this one. 

> So that explains it. 

Yup! :)

closed 2021-03-08 by aallan

fivdi, 2021-04-01:
I think there is actually an issue that might lead people to believe that 1 byte writes don't work. With pico-sdk v1.1, calling `i2c_write_blocking` to write a single byte will always return the value 1, indicating success, even if there is no device at the `addr` passed to `i2c_write_blocking`. See [here](https://github.com/raspberrypi/pico-sdk/issues/278).