raspberrypi/pico-sdk #242: EXECCTRL_JMP_PIN referenced but not explained

dghoti, 2021-03-08:
"EXECCTRL_JMP_PIN" is refernced 3 times in the SDK without an explanation of how/where to set it.
[ https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf page 60 ]

I did eventually find: static void sm_config_set_jmp_pin (pio_sm_config *c, uint pin) but it is a different call and the two are not linked.
[https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf page 149, 153 ]

maybe explain "EXECCTRL_JMP_PIN" better, or change the reference to "sm_config_set_jmp_pin"

lurch, 2021-03-08:
`EXECCTRL_JMP_PIN` is the `JMP_PIN` field of the `SM0_EXECCTRL`, `SM1_EXECCTRL`, `SM2_EXECCTRL` & `SM3_EXECCTRL` Registers described on page 395 of the [RP2040 Datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) .
Unfortunately the fields / registers aren't always referenced in exactly the same manner throughout our documentation, but it sounds like in this particular case if you searched for "jmp_pin" you would have found what you were looking for?

dghoti, 2021-03-08:
I did search for jmp_pin which is how I found the example I quoted above - as a complete beginner in programming and my first attempts at using the pico, I found the reference for this in the SDK very confusing, until I found (and guessed) that it could be set with the sm_config_set_jmp_pin() function.
The description on page 395 of the RP2040 datasheet is even more confusing for me - I would have preferred if the mention of it in page 59 of the SDK had shown how to set or use it, or even referenced the function on 148/152.
No need to go to the RP2040 datasheet, as it is already in the SDK (which is the document that referenced it)

lurch, 2021-03-08:
As "a complete beginner in programming", PIO probably isn't your best starting point!! It's a fairly advanced topic.
And if you don't reference the RP2040 datasheet at the same time as the SDK documentation, I suspect you may find the SDK documentation even harder to understand.

Complete beginners are much better served by MicroPython https://www.raspberrypi.org/documentation/rp2040/getting-started/#getting-started-with-micropython which also lets you use PIO if you want. (Although unfortunately the MicroPython documentation still has gaps https://github.com/micropython/micropython/issues/6855 )

dghoti, 2021-03-08:
I was merely trying to provide some feedback.
I have managed to figure it out (as stated above) but it too me a few hours as the SDK mention of it was rather obscure in my opinion.
maybe "complete beginner" is the wrong phrase, I do know my way around C a bit, which is why I prefer to use C over Python - which while I know is a popular language, I tend to prefer to avoid where possible.
I do realise I am taking the harder path here, but I do prefer to stick to languages with which I am more familiar - the PIO assembly is giving me a hard time, but at least I can follow it and (to a degree) bend it to my purposes, though some others might not be as tenacious as myself - My belief was that the SDK was there to provide a reference, but to only mention setting the jump pin once, and not tie it to the part of the SDK that shows *how* seems to be a limitation.
I am only trying to provide some feedback that might help provide clarity for others - since I have (eventually) found my own solution - which I had mentioned in the very first post.
I guess next time, Ill just keep my mouth shut - apologies if I have caused any offense by my suggestions - I was simply trying to help.
Thank you for your comments lurch, but I dont feel they were helpful or particuarly relevant to the issue I thought I was highlighting,

lurch, 2021-03-09:
Hello @dghoti 

I'm sorry if my reply came across a bit blunt last night, perhaps I was too tired and being overly-defensive :slightly_frowning_face: I'm afraid your "complete beginner in programming" threw me a bit!

We **do** welcome feedback and suggestions, thank you for pointing out which parts of our documentation you found confusing / lacking :+1:

dghoti, 2021-03-09:
Hi @lurch,

Its all good - maybe I understate my programming abilities as I still consider myself an amatuer (compared to a lot of what I see here). Mistakes were made on my side too - you're a good egg!