raspberrypi/pico-sdk #248: Why is the most basic output function the uart hidden?

albertvanderhorst, 2021-03-11:
The most basic output to the uart is the equivalent of the unix write call, accepting a 
buffer and a length. The file is implicit.
This is the function 

**/src/rp2_common/pico_stdio_uart/stdio_uart.c:static void stdio_uart_out_chars(const char *buf, int length) {**

This function is made static (c-ism for private), i.e. invisible to me. 
What reasoning is there behind making the most basic output to uart invisible and  hence unusable?

Wren6991, 2021-03-11:
For direct UART access, you are looking for the `hardware_uart` library, specifically [uart_write_blocking()](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__uart.html#gada6a512414b1113a59725ddfbe23c4a5).

The fact the `stdio_uart` implementation contains a similar function is an implementation detail, which is why it's static

closed 2021-03-11 by Wren6991

albertvanderhorst, 2021-03-12:
My apologies, this is a question that should have been asked on a forum.