raspberrypi/pico-sdk #253: Assembly error with the SVC instruction not supported

dcabanis, 2021-03-13:
The SCV instructions is optionally implemented in some of the Cortex-M0+ (synthesis option) consequently the arm_none_eabi_gcc has to target options one including the SVC instruction: -march=armv6s-m and one without it: -march=armv6-m. As far as I can see the makefile uses -march=armv6-m by default thereby causing the fault: "context_switch.s:163: Error: SVC is not permitted on this architecture". 

the toolchain configuration provided in the pico-sdk. if you navigate to the file: pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake around the line 55 you will find: 

set(ARM_GCC_COMMON_FLAGS " -march=armv6-m -mcpu=cortex-m0plus -mthumb")

This is incorrect; The rp2040 implementation of the Cortex-M0+ includes the SVC instruction. This architecture doesn't. You will need to change the line to the following one: 

set(ARM_GCC_COMMON_FLAGS " -march=armv6s-m -mcpu=cortex-m0plus -mthumb")

lurch, 2021-03-13:
Is this relevant? https://community.arm.com/developer/tools-software/tools/b/tools-software-ides-blog/posts/compiler-flags-across-architectures-march-mtune-and-mcpu
(although it's focused at the HPC end of the spectrum rather than the embedded end!) It says:
### Compiling for Arm
**TL;DR: Whenever possible use only -mcpu=native. Avoid -march and -mtune.**

Does that work for your use-case @dcabanis ? (i.e. does it work if you remove the `-march` option)

Wren6991, 2021-03-17:
Fixed by #260

closed 2021-03-17 by Wren6991