raspberrypi/pico-sdk #256: Add ability to call user function from panic

DusterTheFirst, 2021-03-14:
Panics are, of course, fatal errors, so it makes sense to halt the processor, but a lot of the time, panics are unexpected when they happen (They should get caught or prevented before actual deployment, but things always have a way of going not as planned) and leave the pico in an undefined state. Since the pico does not shut down and stays in the state it was before the panic, it can be useful to reset the pico's outputs or provide other functionality to safely enter the panicked state. 

I have little expertise with C and all its linker magic/tools, but I was envisioning a feature much like the `#[panic_handler]` attribute of rust that would be called in place of the provided panic function.

If achieving this with a macro or some linker tomfoolery is not feasible or elegant, it could also be possible to let the user provide a function pointer that will be called first thing the panic function is entered. Something like:
```c
void (*panic_handler)(const char *fmt, ...);

void __attribute__((noreturn)) __printflike(1, 0) panic(const char *fmt, ...) {
    if (panic_handler) {
        panic_handler(/* pass it the args  */);
    }

    // usual panic code
}
```

kilograham, 2021-03-14:
yup, we should implement this... it's been on my list for a while, but not quite made it!

Lyle-Alloy, 2021-04-06:
I actually had this same thought, but I was thinking it would be easiest to make the panic handler just be a `__weak` (or `__attribute__((weak))`) declaration. 

That way the user can implement their own panic function that doesn't even do any of the usual panic code, which would be desirable for some.

kilograham, 2021-05-04:
> I actually had this same thought, but I was thinking it would be easiest to make the panic handler just be a `__weak` (or `__attribute__((weak))`) declaration.
>
> That way the user can implement their [...]

Right now we only _support_ gcc but with a mind to the use of clang, can someone confirm whether weak works with arm-none-eabi clang? I seem to recall trying it in the past, and it not working, but i don't quite remember.

kilograham, 2021-05-13:
Note also, there should be a trivial way to add a breakpoint at the start of the function via the build which is more useful when debugging

kilograham, 2021-06-02:
merged

closed 2021-06-02 by kilograham