raspberrypi/pico-sdk #257: USB stdio & SDK 1.1.0

Wiz-IO, 2021-03-15:
maybe is issues...

pico_stdio_usb\reset_interface.c:63:32: error: '**SRAM_END**' undeclared (first use in this function);
#include "hardware/regs/addressmap.h"

pico_stdio_usb\stdio_usb.c:17:8: error: unknown type name '**mutex_t**'
#include "pico/mutex.h" 

pico_stdio_usb\stdio_usb.c:19:13: warning: function declaration isn't a prototype [-Wstrict-prototypes]
static void low_priority_worker_irq()

lurch, 2021-03-15:
> pico_stdio_usb\reset_interface.c:63:32: error: 'SRAM_END' undeclared (first use in this function);

@kilograham the offending line says:
```
watchdog_reboot(0, SRAM_END, PICO_STDIO_USB_RESET_RESET_TO_FLASH_DELAY_MS);
```
but the doxygen for [watchdog_reboot](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__watchdog.html#gaa73ccd43d6d755817e71bc3d7eb1fda6) says "If `pc` is non-zero, this will be the stack pointer used.", but here `pc` **is** zero, so that would seem to imply that `SRAM_END` isn't needed? :man_shrugging: 

@Wiz-IO Just out of curiosity, what compiler / compiler-version / compiler-options are you using to catch those errors, which we seem to have missed ourselves? :slightly_smiling_face:

Wiz-IO, 2021-03-15:
Hi,

I make "experimental" platform for PlatformIO & Arduino, based on pico-sdk v1.1.0 and one last version as "backup" (v1.0.1)
Builder: https://github.com/Wiz-IO/wizio-pico
Sources: https://github.com/Wiz-IO/framework-wizio-pico

The compiler is  arm-none-eabi 7.2.1  
support Windows, Linux and OSX ... ( PlatformIO download this version ( pack 1.70201.0 ) auto @ PC system )
and is tested 4 years from me with ARM-s
GCC link: http://dl.platformio.org/packages/manifest.json ( search version gccarmnoneeabi 1.70201.0 @ system )

-Wall 
-Wfatal-errors
-Wstrict-prototypes
-Wno-... will remove later

All default flags is here: https://github.com/Wiz-IO/wizio-pico/blob/main/builder/frameworks/common.py#L104

kilograham, 2021-03-15:
> pico_stdio_usb\stdio_usb.c:19:13: warning: function declaration isn't a prototype [-Wstrict-prototypes]
> static void low_priority_worker_irq()

Yup; this isn't in our standard warning check since TinyUSB has warnings. Will fix

> pico_stdio_usb\reset_interface.c:63:32: error: '**SRAM_END**' undeclared (first use in this function);
> #include "hardware/regs/addressmap.h"
> 
> pico_stdio_usb\stdio_usb.c:17:8: error: unknown type name '**mutex_t**'
> #include "pico/mutex.h"

Agree these should be in place, though for it to fail it implies that you have tinyusb configured not to use RP2040 target, 'cause otherwise they'd both be included via "tusb.h"

Wiz-IO, 2021-03-15:
by the way ... Windows 10 and USB stdio
1. "HelloWorld" example with USB stdio from SDK 1.0.1 - PC driver is OK - USB stdio work
2. same example compiled with SDK 1.1.0 - PC driver unknow ( Zadig can not see the device ) - **USB stdio not work**
3. **#define PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE 0** ( disable reset ) 
4. compile with SDK 1.1.0 - USB stdio is OK :)

lurch, 2021-03-17:
@Wiz-IO That's interesting, thanks.

@kilograham I see that e.g. https://github.com/raspberrypi/tinyusb/blob/master/examples/device/hid_generic_inout/src/usb_descriptors.c#L28 says
"A combination of interfaces must have a unique product id, since PC will save device driver after the first plug.
Same VID/PID with different interface e.g MSC (first), then CDC (later) will possibly cause system error on PC."
Could that explain the behaviour that @Wiz-IO describes in his last message? If so, does that imply that "USB-CDC + USB_ENABLE_RESET" needs a different PID from just "USB-CDC"?

kilograham, 2021-03-17:
that could certainly be it. @Wiz-IO can you try changing the line 35 in src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c

```
#define USBD_PID (0x000a) // Raspberry Pi Pico SDK CDC
```

to something else maybe `0xeeea` and see if Windows likes that

Wiz-IO, 2021-03-17:
reset enabled
[image]

reset disabled
[image]

kilograham, 2021-03-17:
was that with zadig? i.e. did zadig not like the "new" device?

Wiz-IO, 2021-03-17:
the driver is installed ... but win 10 work without zadig ( I do not remember )

enable/disable reset is
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE **1/0**
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L72

Wiz-IO, 2021-03-17:
btw: win 7 - unknow device for both ( driver is for SDK 1.0.1 ) - I need to delete from registry driver uuid and restart PC 
( will test tomorrow )

lurch, 2021-03-18:
Yes, Windows 7 requires extra steps that Windows 10 doesn't - the [getting started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) has a link to https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=300053

kilograham, 2021-03-18:
I'm super confused about what does and doesn't work now; does Windows 10 work correctly with a not seen before (different PID) device which uses the reset interface (i.e. is composite)?

Wiz-IO, 2021-03-18:
#define USBD_PID (0x000a) up driver COM4
#define USBD_PID (0xeeea) up driver COM5

someone other to test Reset vendor on Win10 - it doesn't work for me - Unknown device...

kilograham, 2021-03-27:
@lurch can you take a look at this to see what does and doesn't work?

lurch, 2021-03-27:
@kilograham I'm afraid all my computers run Linux, I don't have anything with Windows 10 :-/  I think @liamfraser uses Windows though?

Wiz-IO, 2021-03-28:
**RESET_0/1**.TXT mean PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE **0/1**
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/SDK/include/pico/stdio_usb.h#L72

logs form WIN10
https://github.com/Wiz-IO/TEST

kilograham, 2021-03-28:
OK, thanks that is helpful

liamfraser, 2021-03-30:
@Wiz-IO if you open Zadig and List all devices what is the driver for the Board CDC interface with the standard VID + PID. It should be usbser:

[image]

Wiz-IO, 2021-03-30:
my WIN-10 work without Zadig

btw: here it tries to load a second driver
https://github.com/Wiz-IO/TEST/blob/main/RESET_1.TXT#L88

kilograham, 2021-03-30:
Can you please explicitly state exactly what *does and does not* work

You said higher above that Windows 10 does not work for you, and then you say it does work

Wiz-IO, 2021-03-30:
not work reset interface
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE = 1

Wiz-IO, 2021-03-30:
IF PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE = 1 windows say unknown driver... if = 0 the driver is ok, stdio work...
sorry... i write from phone

lurch, 2021-03-30:
The reset interface won't work because it's a "custom interface" which we don't supply a Windows-driver for yet.
But the USB stdio interface should still work with the RESET_VIA_VENDOR_INTERFACE enabled?

Wiz-IO, 2021-03-30:
it doesn't work for me - Test it...

I have disabled it for PlatformIO port
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/SDK/include/pico/stdio_usb.h#L72

lurch, 2021-03-30:
As asked above, can you please show a Zadig screenshot for the Board CDC interface? You might need to enable the "list all devices" option or something.

Wiz-IO, 2021-03-30:
I posted it...
https://github.com/raspberrypi/pico-sdk/issues/257#issuecomment-801237579

kilograham, 2021-03-30:
that is not what was asked for, and yes we've obviously tested it and it works fine, so we're trying to help figure out why it doesn't work on *your* machine

Wiz-IO, 2021-03-30:
WIN 7
[image]

Wiz-IO, 2021-03-30:
**no reset**  - WIN 7 ( 10 work too )

[image]

kilograham, 2021-04-01:
> WIN 7
> [image]

can you close the dropdown so we can see what is underneath it for the reset case (show the results for both CDC interfaces)

Wiz-IO, 2021-04-01:
the USB not exist in this menu

kilograham, 2021-04-01:
what you have selected right there "USB Device Interface 0" and "USB Device Interface 1"

Wiz-IO, 2021-04-01:
other USB drivers
[image]

kilograham, 2021-04-01:
are you able to switch that to WinUSB?

Wiz-IO, 2021-04-01:
Pico as WinUSB? without reset interface is OK

shot is from WIN 10
[image]

kilograham, 2021-04-01:
no; are you able to switch libusbK to WinUSB on the _broken_ versions?

Wiz-IO, 2021-04-01:
no driver...
[image]

Wiz-IO, 2021-04-01:
can you paste me compilator verbose log ( part ) 
with PICO_STDIO_USB and PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE = 1

lurch, 2021-04-01:
http://www.linux-usb.org/usb.ids says that 09DA:9066 is "A4Tech Co., Ltd. F3 V-Track Gaming Mouse".

Wiz-IO, 2021-04-01:
yep, it is my mouse :)

closed 2021-04-01 by Wiz-IO