raspberrypi/pico-sdk #258: Build option PICO_TOOLCHAIN_PATH has no effect

josesimoes, 2021-03-15:
Was trying to use `PICO_TOOLCHAIN_PATH` to point to the location where I have it installed on my setup.
(I do _NOT_ have gcc it on my path)

Turns out that this option has no effect because it's always cleared [here](https://github.com/raspberrypi/pico-sdk/blob/26653ea81e340cacee55025d110c3e014a252a87/cmake/pico_pre_load_toolchain.cmake#L3).

As soon as I comment out this line, I can set the `PICO_TOOLCHAIN_PATH` option and `pico_find_compiler` happilly finds the compiler.

I can submit a PR with this fix if you want.

lurch, 2021-03-15:
Probably better to add a `if (NOT DEFINED PICO_TOOLCHAIN_PATH)` around it? :shrug: (that's only a guess, I know very little about CMake)

kilograham, 2021-03-15:
how are you setting `PICO_TOOLCHAIN_PATH`? was this a fresh build directory?

josesimoes, 2021-03-15:
> how are you setting `PICO_TOOLCHAIN_PATH`? was this a fresh build directory?

Yes, it's a fresh build directory and I'm using VS Code ".vscode\cmake-variants.json" to set the build options.

kilograham, 2021-03-15:
what i meant was how is it getting set from a cmake perspective? command line define, environment?

I know it works with `-DPICO_TOOLCHAIN_PATH=` on the command line, since i use that all the time!

josesimoes, 2021-03-15:
I was using cmake-variants.json to build with CMake with the [CMake Tools](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools) extension.

josesimoes, 2021-03-15:
@kilograham do you use _just_ `-DPICO_TOOLCHAIN_PATH=`  or you actually have _something_ in the path ahead of the `=` ?

lurch, 2021-03-15:
I _suspect_ Graham meant `-DPICO_TOOLCHAIN_PATH=/path/to/toolchain`

kilograham, 2021-03-15:
yes, that is what I meant... also what OS are you on?

josesimoes, 2021-03-15:
Windows 10, 64bits
VS Code 1.54.1
CMake tools 1.6.0

josesimoes, 2021-03-16:
Follows the CMake logs, for completeness. On both the `PICO_TOOLCHAIN_PATH` is set trough `.vscode\cmake-variants.json`. Build folder cleaned.

With the original code:

```
[main] Configuring folder: pico-sdk 
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --no-warn-unused-cli -DCMAKE_MAKE_PROGRAM:STRING=E:/nftools/Ninja/ninja.exe -DPICO_TOOLCHAIN_PATH:STRING=E:/nftools/GNU_Tools_ARM_Embedded/9-2020-q2-update/ -DPICO_COMPILER:STRING=pico_arm_gcc -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -He:/Github/pico-sdk -Be:/Github/pico-sdk/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] PICO_SDK_PATH is E:/Github/pico-sdk
[cmake] Defaulting PICO_PLATFORM to rp2040 since not specified.
[cmake] PICO compiler is pico_arm_gcc
[cmake] PICO_GCC_TRIPLE defaulted to arm-none-eabi
[cmake] CMake Error at cmake/preload/toolchains/find_compiler.cmake:28 (message):
[cmake]   Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
[cmake]   "PICO_TOOLCHAIN_PATH".
[cmake] Call Stack (most recent call first):
[cmake]   cmake/preload/toolchains/pico_arm_gcc.cmake:20 (pico_find_compiler)
[cmake]   C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
[cmake]   CMakeLists.txt:6 (project)
[cmake] 
[cmake] 
[cmake] CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
[cmake] CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
[cmake] CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
[cmake] -- Configuring incomplete, errors occurred!
```

With line 3 @ pico_pre_load_toolchain.cmake commented:

```
[cmake] -- Build files have been written to: E:/Github/pico-sdk/build
[main] Configuring folder: pico-sdk 
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --no-warn-unused-cli -DCMAKE_MAKE_PROGRAM:STRING=E:/nftools/Ninja/ninja.exe -DPICO_TOOLCHAIN_PATH:STRING=E:/nftools/GNU_Tools_ARM_Embedded/9-2020-q2-update/ -DPICO_COMPILER:STRING=pico_arm_gcc -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -He:/Github/pico-sdk -Be:/Github/pico-sdk/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] PICO_SDK_PATH is E:/Github/pico-sdk
[cmake] Defaulting PICO_PLATFORM to rp2040 since not specified.
[cmake] PICO compiler is pico_arm_gcc
[cmake] PICO_GCC_TRIPLE defaulted to arm-none-eabi
[cmake] -- The C compiler identification is GNU 9.3.1
[cmake] -- The CXX compiler identification is GNU 9.3.1
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: E:/nftools/GNU_Tools_ARM_Embedded/9-2020-q2-update/bin/arm-none-eabi-gcc.exe
[cmake] Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
[cmake] Defaulting PICO target board to pico since not specified.
[cmake] Using board configuration from E:/Github/pico-sdk/src/boards/include/boards/pico.h
[cmake] -- Found Python3: C:/Program Files/Python39/python.exe (found version "3.9.2") found components:  Interpreter 
[cmake] TinyUSB available at E:/Github/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
[cmake] Compiling TinyUSB with CFG_TUSB_DEBUG=1
[cmake] ELF2UF2 will need to be built
[cmake] -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: E:/Github/pico-sdk/build
```

kilograham, 2021-03-16:
Ah, well `-DPICO_TOOLCHAIN_PATH:STRING=...` wasn't something i was even aware of, and is the reason for the difference in behavior.

Can you check

```
set(PICO_TOOLCHAIN_PATH "${PICO_TOOLCHAIN_PATH}" CACHE INTERNAL "")
```

works for you

josesimoes, 2021-03-16:
Yes, that works. 

🤔  I didn't think of adding the `"` because, from the comment, I was under the impression that the intent of the line was precisely to clear the variable.

josesimoes, 2021-03-16:
I can still submit the PR with your fix, if you want to. Not sure if you prefer to handle this other way...
Just let me know.

kilograham, 2021-03-16:
sure, thx

kilograham, 2021-03-22:
merged into 1.1.1

jbhntr861, 2022-12-07:
Hello I am having this issue I believe

"CMake Warning at pico-sdk/cmake/preload/toolchains/find_compiler.cmake:22 (message):
  PICO_TOOLCHAIN_PATH specified (/c/Program Files
  (x86)/ARMgnu/mpac/bin/arm-none-eabi-gcc), but arm-none-eabi-gcc not found
  there
"

I have tried a million different ways of setting the path to arm-none-eabi-gcc and it will not find it there. Is there any solution to this?

peterharperuk, 2022-12-07:
You're commenting on an issue that was fixed some time ago. You might have more luck raising a new issue if you think you have found a new problem. If you just want support - try the forum https://forums.raspberrypi.com/viewforum.php?f=144

PR #262 (2021-03-18): Fix loading of PICO_TOOLCHAIN_PATH
- Add double quotes because build option it's a string.
- Remove comment as requested by @kilograham.
- Resolves #258.
commit 304ab7dd (2021-03-18): Fix loading of PICO_TOOLCHAIN_PATH (#262)

- Add double quotes because build option it's a string.
- Remove comment as requested by @kilograham.
- Resolves #258.