raspberrypi/pico-sdk #264: Can't malloc the 4KB of the heap

MichaelBell, 2021-03-17:
malloc won't allocate the last 4KB of available heap (addresses in 2003F000- 20040000).  A simple test program is here: https://gist.github.com/MichaelBell/a13513a483a07a41c5b74e8458fba51d

I believe the problem is _sbrk should allow heap_end to be moved up to __StackLimit.  I've got a [commit which appears to fix the issue](https://github.com/MichaelBell/pico-sdk/commit/4d84592bef7cdeb2b1f4081ca528875c14fa447b) but I'm not sufficiently familiar wit hhow this is meant to work to know whether that's the correct approach.

kilograham, 2021-03-18:
yes your fix looks sensible... it is all a bit of a hack (the OPTIMISTIC_SBRK thing) since SBRK is being called for 4K but you are not giving it 4K...

MichaelBell, 2021-03-18:
Cool, I've made a pull request with a brief explanation of what it does.

kilograham, 2021-03-22:
can you please rebase onto develop-1.1.1

MichaelBell, 2021-03-22:
I've rebased and updated the PR.

kilograham, 2021-03-24:
merged

closed 2021-03-24 by kilograham