raspberrypi/pico-sdk #265: No function for gpio irq over

Volksolive, 2021-03-17:
There is functions to invert the signal for gpio out, in and out enable (oe) but there is no function for the gpio irq signal invert, something like gpio_set_irqover to stay consistent with the other functions.
Maybe it could also be nice in the gpio_set_inover to add a third parameter like bool irqover which would invert both in and irq if true or only in if false.
I can see cases where you would want both to be inverted, for example a debounce/filter where you want the IT to start a delay and after this delay, check the state of the pin.

Volksolive, 2021-03-18:
BTW the gpio_override enum is not mentioned in the SDK documentation

kilograham, 2021-04-08:
Same bucket as #290

PR #470 (2021-06-03): Add gpio_set_irqover to match inover/outover/oeover (fixes #265)
fixes #265
commit bba0420c (2021-06-03): Add gpio_set_irqover to match inover/outover/oeover (fixes #265) (#470)