raspberrypi/pico-sdk #276: Recursive mutex

Wiz-IO, 2021-03-22:
Please add support for **recursive mutex**
mutex_enter_blocking_recursive()
mutex_try_enter_recursive()
mutex_exit_recursive()

kilograham, 2021-03-24:
So the existing mutex classes should not be used in a FreeRTOS as they are core not task focused (unless you specifically mean core).

Equally, should a recursive mutex be successfully claimed by both application and IRQ on the same core?

kilograham, 2021-03-25:
showing me that code doesn't really answer my question, though I am able to glean that you seem to want the recursive mutexes to work the way that makes them suitable for using in newlib locking (which isn't documented there).

Can you point me at some docs for that - since you seem to be implementing the locks defined somewhere, because fixing enabling newlib re-entrancy/locking when using multicore via the raw SDK is likely a good thing anyway (note I was assuming the newlib that comes with gcc just didn't support it, but I guess I am wrong).

Wiz-IO, 2021-03-25:
I was looking for documents for newlib
found only this...
https://www.cs.ccu.edu.tw/~pahsiung/courses/ese/resources/newlib.pdf

and I applied this in reverse 
https://gist.github.com/thomask77/65591d78070ace68885d3fd05cdebe3a

here is my version (as idea)
is only one counter in mutex_t and this
https://github.com/Wiz-IO/framework-wizio-pico/blob/main/SDK/pico/pico_sync/mutex.c#L82

kilograham, 2021-05-04:
Note recursive mutexes will be added by #378

lurch, 2021-05-06:
...**have** been added... :slightly_smiling_face: (i.e. this can be closed?)

Wiz-IO, 2021-05-06:
yep close
this days will test...

closed 2021-06-24 by kilograham