raspberrypi/pico-sdk #277: ELF _entry_point function does not seem to work as expected

aggarg, 2021-03-22:
The purpose of the following code snippet in crt0.S seems to allow an external debugger to start running form the ELF entry point after loading the image to the external flash:

```c
// ELF entry point:
.type _entry_point,%function
.thumb_func
.global _entry_point
_entry_point:

#if PICO_NO_FLASH
    // Vector through our own table (SP, VTOR will not have been set up at
    // this point). Same path for debugger entry and bootloader entry.
    ldr r0, =__vectors
#else
    // Debugger tried to run code after loading, so SSI is in 03h-only mode.
    // Go back through bootrom + boot2 to properly initialise flash.
    movs r0, #0
#endif
    ldr r1, =(PPB_BASE + M0PLUS_VTOR_OFFSET)
    str r0, [r1]
    ldmia r0!, {r1, r2}
    msr msp, r1
    bx r2
```

This, however, does not seem to work in the Segger Ozone. `ldmia r0!, {r1, r2}` and `msr msp, r1` seems to make no difference to `r1`, `r2` or `msp`. What can be the reason? Can it be that the bootrom and second stage bootloader have not yet configured the flash for XIP and we are trying to execute this entry point in flash?

The work around I used is to update the default generated Segger Ozone Jdebug file to read SP and PC from 0x00000000, thereby making it start from bootrom. 

Thanks.

aggarg, 2021-03-22:
In fact, a better way to do the same is in Ozone is to leave the functions `AfterTargetDownload()` and `AfterTargetReset()` in the jdebug file empty which causes the rom bootloader to run: https://wiki.segger.com/Debug_on_a_Target_with_Bootloader

The question then is - is this entry_point needed?

Thanks.

kilograham, 2021-03-22:
> The work around I used is to update the default generated Segger Ozone Jdebug file to read SP and PC from 0x00000000, thereby making it start from bootrom.

That is exactly what happens unless `PICO_NO_FLASH=1` as per the comment: 
```
// Debugger tried to run code after loading, so SSI is in 03h-only mode.
// Go back through bootrom + boot2 to properly initialise flash.
````

The code explicitly reads SP and PC from 0 and jumps to it, you say "seems to make no difference to r1, r2 or msp"; what were these before and after then?

Your debugger is not _required_ to jump to the entry point, but it is quite a sensible way!

aggarg, 2021-03-23:
> The code explicitly reads SP and PC from 0 and jumps to it, you say "seems to make no difference to r1, r2 or msp"; what were these before and after then?

```
r0 = 0
r1 = E000 ED08
r2 = FFFF FFFF
msp = 4B32 B500
```

`ldmia r0!, {r1, r2}` and `msr msp, r1` seems to make no change to r0, r1, r2 or msp. 

> Your debugger is not required to jump to the entry point, but it is quite a sensible way!

My experience is that it is required as the `_entry_point` code does not work. Which debugger does this code work with? 

Thanks.

kilograham, 2021-03-23:
> Can it be that the bootrom and second stage bootloader have not yet configured the flash for XIP and we are trying to execute this entry point in flash?

Ah i missed this... yes, this could be the problem (i.e. garbage is being executed)

this works with (IDE->)gdb->openocd as the flash code for rp2040 in our openocd puts the flash back in generic XIP mode after wiritng to the flash.

How is Ozone writing code to the flash?

kilograham, 2021-03-23:
Note that we use the entry point by default under debug, since we support non-flash binaries also (i.e. loading directly into RAM). 

You can certainly start using the bootrom vector (which is exactly what the flash entry point does anyway) which is what i meant above, but `gdb load foo.elf; c` is a handy way to debug too.

aggarg, 2021-03-23:
> How is Ozone writing code to the flash?

I am not sure - this is internal to Ozone. Well, as long as this code works for some setup [gdb --> OpenOCD], it is all good.

> Note that we use the entry point by default under debug, since we support non-flash binaries also (i.e. loading directly into RAM).

Yes, good point. 

Thank you for all the information. I will close this issue.

Thanks.

closed 2021-03-23 by aggarg

lurch, 2021-03-24:
This issue is mostly over my head, but could a `#ifdef DEBUGGER_DOESNT_REENABLE_XIP` code-path in crt0.S make sense @kilograham ?