raspberrypi/pico-sdk #278: i2c_write_blocking not returning PICO_ERROR_GENERIC when expected

fivdi, 2021-03-22:
The program below is a variant of [bus_scan](https://github.com/raspberrypi/pico-examples/blob/sdk-1.1.0/i2c/bus_scan/bus_scan.c) from pico-examples. It calls ```i2c_write_blocking``` rather than ```i2c_read_blocking```.

```c
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/i2c.h"

static bool reserved_addr(uint8_t addr) {
    return (addr & 0x78) == 0 || (addr & 0x78) == 0x78;
}

int main() {
    stdio_init_all();

    i2c_init(i2c_default, 100 * 1000);
    gpio_set_function(PICO_DEFAULT_I2C_SDA_PIN, GPIO_FUNC_I2C);
    gpio_set_function(PICO_DEFAULT_I2C_SCL_PIN, GPIO_FUNC_I2C);
    gpio_pull_up(PICO_DEFAULT_I2C_SDA_PIN);
    gpio_pull_up(PICO_DEFAULT_I2C_SCL_PIN);

    printf("\ni2c_write_blocking test\n");
    printf("   0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F\n");

    for (int addr = 0; addr < (1 << 7); ++addr) {
        if (addr % 16 == 0) {
            printf("%02x ", addr);
        }

        int ret;
        uint8_t txdata = 0;
        if (reserved_addr(addr)) {
            ret = PICO_ERROR_GENERIC;
        } else {
            ret = i2c_write_blocking(i2c_default, addr, &txdata, 1, false);
            if (ret != 1) {
               printf("never printed as ret is always 1\n");
            }
        }

        printf(ret < 0 ? "." : "@");
        printf(addr % 16 == 15 ? "\n" : "  ");
    }
}
```

Although there is only one device on the I2C bus, an MCP9808 temperature sensor at address 0x18, the output of this program is:

```
i2c_write_blocking test
   0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00 .  .  .  .  .  .  .  .  @  @  @  @  @  @  @  @
10 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
20 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
30 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
40 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
50 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
60 @  @  @  @  @  @  @  @  @  @  @  @  @  @  @  @
70 @  @  @  @  @  @  @  @  .  .  .  .  .  .  .  .
```

```i2c_write_blocking``` returns 1 every time it's called although it should only return 1 when called for `addr` 0x18. ```i2c_write_blocking``` should return `PICO_ERROR_GENERIC` every time it's called except when called for `addr` 0x18.

fivdi, 2021-03-27:
I did some further testing here and this issue appears to be related to the following [block of code](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_i2c/i2c.c#L140-L150) in function `i2c_write_blocking_internal()`:

```c
        do {
            // Note clearing the abort flag also clears the reason, and this
            // instance of flag is clear-on-read!
            abort_reason = i2c->hw->tx_abrt_source;
            abort = (bool) i2c->hw->clr_tx_abrt;
            if (timeout_check) {
                timeout = timeout_check(ts);
                abort |= timeout;
            }
            tight_loop_contents();
        } while (!abort && !(i2c->hw->status & I2C_IC_STATUS_TFE_BITS));
```

I think there are three issues with this block of code.

**Issue 1**

```c
            abort_reason = i2c->hw->tx_abrt_source;
            abort = (bool) i2c->hw->clr_tx_abrt;
```

The first line of code here copies the value of the `IC_TX_ABRT_SOURCE` register to `abort_reason`. The second line clears the same register, that is, the `IC_TX_ABRT_SOURCE` register. If the hardware decides to set a bit in the `IC_TX_ABRT_SOURCE` register after the first line has been executed, but before the second line is executed, the reason for the aborted transfer will be lost.

**Issue 2**

```c
            abort = (bool) i2c->hw->clr_tx_abrt;
```

Reading the `IC_CLR_TX_ABRT` register clears the `IC_TX_ABRT_SOURCE` register, which is the correct thing to do here, however, the value of the `IC_CLR_TX_ABRT` register is always 0. This means that the value of `abort` will always be `false` and `abort` will not serve the purpose it is intended to serve. Note that the RP2040 Datasheet does not say that the  `IC_CLR_TX_ABRT` should ever have a value other than 0.

**Issue 3**

```c
        } while (!abort && !(i2c->hw->status & I2C_IC_STATUS_TFE_BITS));
```

The condition in the do-while loop allows the loop to continue execution if `abort` is `false` and the transmit FIFO is not empty. This condition is acceptable for all bytes being transmitted except the last byte. When the last byte is being transmitted, the transmit FIFO will be empty before the byte has actually been sent over the wire. The loop will complete execution too early when the last byte is being transmitted. In reality, the `i2c_write_blocking_internal()` function will have returned before the I2C peripheral has finished transmitting the last byte. In addition, `i2c_write_blocking_internal()` will return a value indicating that it has run successfully although this is not the case.

My first attempt to fix all three issues is the following:
```c
        do {
            // Note clearing the abort flag also clears the reason, and this
            // instance of flag is clear-on-read!
            abort_reason = i2c->hw->tx_abrt_source;
            if (abort_reason) {
                i2c->hw->clr_tx_abrt;
                abort = true;
            }
            if (timeout_check) {
                timeout = timeout_check(ts);
                abort |= timeout;
            }
            tight_loop_contents();
        } while (!abort &&
                 (!(i2c->hw->status & I2C_IC_STATUS_TFE_BITS) ||
                   (last && (i2c->hw->status & I2C_IC_STATUS_MST_ACTIVITY_BITS))));
```

Unfortunately, this only works when the value of the `nostop` argument passed to `i2c_write_blocking_internal()` is `false`.

The condition in the do-while loop now allows the loop to continue execution if ((abort is false) && (the transmit FIFO is not empty || there is master activity on the I2C bus)).

When the value of the `nostop` argument passed to `i2c_write_blocking_internal()` is `true`, the master will hold SCL low after transmitting the last byte. When the master holds the SCL low, there is master activity, so the loop turns into an infinite loop.

So, the question is, how can it be determined that the last byte that need to be transmitted has been completely transmitted over the wire?

fivdi, 2021-04-01:
> So, the question is, how can it be determined that the last byte that needs to be transmitted has been completely transmitted over the wire?

What looks promising here is the `MASTER_ON_HOLD` bit in the `IC_RAW_INTR_STAT` register which is documented as follows:

>Indicates whether master is holding the bus and TX FIFO 
>is empty. Enabled only when
>I2C_DYNAMIC_TAR_UPDATE=1 and
>IC_EMPTYFIFO_HOLD_MASTER_EN=1.
>
>Reset value: 0x0
>0x0 → MASTER_ON_HOLD interrupt is inactive
>0x1 → MASTER_ON_HOLD interrupt is active

However, the `MASTER_ON_HOLD` bit never appears to have the value 1.

Is the master `MASTER_ON_HOLD` bit in the `IC_RAW_INTR_STAT` register known to function correctly?

See also this [issue](https://github.com/raspberrypi/pico-feedback/issues/133).

Wren6991, 2021-04-09:
Thanks, have confirmed this issue, and agree that this is caused by exiting the loop in between the FIFO emptying and the NAK taking place. The I2C hardware docs are a bit impenetrable as you say, so I'll take a closer look at the hardware and see if there are any other helpful flags.

Wren6991, 2021-04-09:
So as mentioned on https://github.com/raspberrypi/pico-feedback/issues/133 the `MASTER_ON_HOLD` flag is not present on RP2040, because we don't support update of TAR whilst the bus is held (and this is a docs issue on our end which we need to fix).

`I2C_IC_STATUS_MST_ACTIVITY` looks promising, just need to confirm that the FSM returns to the idle state on NAK:

```
// -----------------------------------------------------------------------------
// Field       : I2C_IC_STATUS_MST_ACTIVITY
// Description : Master FSM Activity Status. When the Master Finite State
//               Machine (FSM) is not in the IDLE state, this bit is set. - 0:
//               Master FSM is in IDLE state so the Master part of DW_apb_i2c is
//               not Active - 1: Master FSM is not in IDLE state so the Master
//               part of DW_apb_i2c is Active Note: IC_STATUS[0]-that is,
//               ACTIVITY bit-is the OR of SLV_ACTIVITY and MST_ACTIVITY bits.
//
//               Reset value: 0x0
//               0x0 -> Master is idle
//               0x1 -> Master not idle
```

Wren6991, 2021-04-09:
> So, the question is, how can it be determined that the last byte that needs to be transmitted has been completely transmitted over the wire?

This is a good question, I have been reading through the master transaction FSM to try and answer it. There is a state the FSM enters upon bottoming out the FIFO during a write transfer, when the STOP bit is not set on the last FIFO record, and this state is distinct from the idle state.

Unfortunately the only way to detect this state I can see at the moment is a set of debug signals, which are not hooked up, so I there is no obvious way of detecting the difference between FIFO-empty-and-clocking and FIFO-empty-and-stalled when no STOP condition is asserted. Will keep reading.

Wren6991, 2021-04-09:
There are some potential hacks using implementation details of the I2C hardware, e.g. disabling restarts mid-transfer and putting a read after the final TX byte should cause the I2C to generate an abort flag after the final TX byte is posted, which can then be detected as a successful completion. May have to resort to something like that if we can't find a flag that disambiguates FIFO empty vs waiting for FIFO data.

lurch, 2021-04-09:
The GPL-vs-BSD argument obviously means we have to be careful not to copy the code, but is it possible to get "inspiration" from the Linux driver mentioned here: https://github.com/raspberrypi/pico-sdk/pull/273#issuecomment-803654867  ? :thinking:

fivdi, 2021-04-09:
I've actually spent quit a bit of time trying to figure this one out. The code that I currently have is a bit of a mess and not worthy of publishing.

Anyway, what I have done so far is to set `TX_EMPTY_CTRL` [here](https://github.com/raspberrypi/pico-sdk/blob/1.1.2/src/rp2_common/hardware_i2c/i2c.c#L42-L46) in `i2c_init`. This results in the `TX_EMPTY` bit in the `IC_RAW_INTR_STAT` being set much later, i.e., when the transmit buffer is at or below the threshold value set in the IC_TX_TL register and the transmission of the address/data from the internal shift register for the most recently popped command is completed.

Then the [do-while loop here](https://github.com/raspberrypi/pico-sdk/blob/1.1.2/src/rp2_common/hardware_i2c/i2c.c#L150) can test `I2C_IC_RAW_INTR_STAT_TX_EMPTY_BITS` in the condition to see when to terminate.

Then after the do-while loop has completed, it's possible to wait for I2C_IC_RAW_INTR_STAT_STOP_DET_BITS to be set if appropriate, i.e., if there was an abort or if it's the last byte to send and `nostop` is false.

This roughly what I have done, but there is more to it than that.

fivdi, 2021-04-11:
#331 should fix this.

fivdi, 2021-05-05:
#331 has been merged so I'll go ahead and close this issue.

PR #331 (2021-05-05): i2c: improve communication with i2c devices in i2c_write_blocking
This PR should fix #278.

For testing, the following three test programs were used:

- [bus_scan_issue_278](https://github.com/fivdi/rp2040-tests/tree/master/i2c/bus_scan_issue_278)
- [scan_at_many_baudrates](https://github.com/fivdi/rp2040-tests/tree/master/i2c/scan_at_many_baudrates)
- [read_and_write_devices](https://github.com/fivdi/rp2040-tests/tree/master/i2c/read_and_write_devices)

Each test program has a readme with a short explanation of their purpose.

~~(While working on this issue I noticed a problem with baudrate 1 [... cut]
closed 2021-05-05 by fivdi