raspberrypi/pico-sdk #283: i2c speed incorrect and non-linear giving significant error for 400kHz

kevinjwalters, 2021-03-25:
I've been testing from CircuitPython but apparently that simply uses SDK `i2c_init` / `i2c_set_baudrate` and @dhalbert confirmed this. 100kHz is 90.9, 400kHz is 266.2kHz. This looks like it's an SDK problem although I glanced at SDK code and couldn't see how it could cause this...

[image]

Very minor possibility my logic analyser is bust. It would be useful to get independent confirmation of this.

I did notice the hardware has two pairs of very similar-sounding registers and the SDK only sets the fast ones.

https://github.com/raspberrypi/pico-sdk/blob/503bc8b385d2e87390b5c7c0a12bc32fa9025944/src/rp2_common/hardware_i2c/i2c.c#L84-L85

https://github.com/raspberrypi/pico-sdk/blob/26653ea81e340cacee55025d110c3e014a252a87/src/rp2040/hardware_structs/include/hardware/structs/i2c.h#L19-L22

Wren6991, 2021-03-25:
Do you have external pullups on the I2C bus? Are you able to capture any analog traces of what is going on? RP2040's internal pullups are quite high-resistance, and a slow rising edge will limit the maximum frequency due to clock stretching.

Wren6991, 2021-03-25:
[image]

I can fit your curve quite well by assuming a 1.17 microsecond rise time, stretching each clock pulse.

kevinjwalters, 2021-03-25:
10k external, I don't have a storage scope :(

Wren6991, 2021-03-25:
Ok. Does your measured frequency become more linear if you bump those to 2k2?

kevinjwalters, 2021-03-25:
It does indeed get faster, I just tried 400k with 10k on device and 2.2k which gives me combined 1.8k plus whatever CircuitPython/SDK might be doing with internal RP2040 and got 352.990 kHz. I didn't realise that pullups would have an effect on intrabyte clock pulses for clock stretching, I need to read more about this.

```
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; Raspberry Pi Pico with rp2040
>>>
>>> import board
>>> import busio
>>> i2c = busio.I2C(scl=board.GP1, sda=board.GP0, frequency=400_000)
>>> i2c.try_lock()
True
>>> buffer = bytearray([0x81, 0x50, 0x81, 0x69])
>>> i2c.writeto(60, buffer)
```

kevinjwalters, 2021-03-25:
What's the implementation here? Is it this sort of thing with just a wait for it to go high and no modification of final delay?

```
set_clock(low)
delay(fs_scl_lcnt)
set_clock(high)
wait_clock(high)
delay(fs_scl_hcnt)
```

kevinjwalters, 2021-03-27:
I tried with 1k pull ups and I can get 96.107kHz  for 100 and 366.569kHz for 400. I had a flick through the spec and it's only high speed mode that prohibts intrabyte clock stretching.

The clock low time is 4.080-4.120us for 100kHz and 1.080-1.120us for 400kHz, shown in trace below. Both look _below the minimum **tlow**_ in the i2c specification? Am I reading Table 10 correctly in https://www.nxp.com/docs/en/user-guide/UM10204.pdf ? Are the `hcnt` and `lcnt` values around the wrong way?

[image]

[image]

dhalbert, 2021-03-28:
> I did notice the RP2040 hardware has two pairs of very similar-sounding registers and the SDK only sets the fast ones.

After studying the datasheet and code, I believe this is explained by this bit in `i2c.c`:
```c
    // Always use "fast" mode (<= 400 kHz, works fine for standard mode too)
    hw_write_masked(&i2c->hw->con,
                   I2C_IC_CON_SPEED_VALUE_FAST << I2C_IC_CON_SPEED_LSB,
                   I2C_IC_CON_SPEED_BITS
    );
    i2c->hw->fs_scl_hcnt = hcnt;
    i2c->hw->fs_scl_lcnt = lcnt;
    i2c->hw->fs_spklen = lcnt < 16 ? 1 : lcnt / 16;
```
The controller is always run in "fast" mode, but the actual speed is set by the `hcnt` and `lcnt` computations.

kevinjwalters, 2021-04-07:
As discussed, this slowdown is caused by the clock stretching implementation.

closed 2021-04-07 by kevinjwalters

hemant-01, 2024-02-05:
@kevinjwalters hi kevin, although its been long time since this issue... i am connecting pico with a sensor([icm-42670-p](https://invensense.tdk.com/products/motion-tracking/6-axis/icm-42670-p/)) by i2c fast mode plus (as this is what sensor supports), from your understanding, would there be any issues in data communication?

kevinjwalters, 2024-02-11:
@hemant-01 I'm not familiar with that particular sensor but I think you'll be fine. This issue was raised just because I didn't understand why the requested clock rate / speed wasn't achieved.

Re-reading comments it's not clear if https://github.com/raspberrypi/pico-sdk/issues/283#issuecomment-808746340 was ever addressed but perhaps that's a very minor issue.