raspberrypi/pico-sdk #285: Flash error using openocd

alexandruradovici, 2021-03-27:
I am trying to flash a binary with the following layout:

```
MEMORY
{
  /* boot from Flash */
  rom (rx)  : ORIGIN = 0x10000000, LENGTH = 128K
  prog (rx) : ORIGIN = 0x10020000, LENGTH = 1920K
  ram (rwx) : ORIGIN = 0x20004000, LENGTH = 256K
}

MPU_MIN_ALIGN = 8K;
PAGE_SIZE = 4K;
```

When trying to flash, I get the following openocd output:

```
$ openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg 
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-02-23-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
Error: Whole bank access must start at beginning of bank.
Error: flash_erase returned -904
```

Flashing works fine if I remove the `prog` line from the layout file or put the address somewhere in RAM (> 0x2000000).

kilograham, 2021-03-31:
when you say "when I try to flash"... how? is this VS Code? seems like it may be a problem with the gdb command being sent

alexandruradovici, 2021-03-31:
I am using gdb with the following sequence of commands:

```
target remote :3333
load
```

kilograham, 2021-03-31:
can you attach your ELF

alexandruradovici, 2021-03-31:
Sure 
[raspberry_pi_pico.zip](https://github.com/raspberrypi/pico-sdk/files/6238550/raspberry_pi_pico.zip)

alexandruradovici, 2021-04-05:
Hi, do you have any update here?

kilograham, 2021-04-08:
Yes, i just took a look, and I note that it doesn't look like your are using the pico-sdk, so this probably belongs in our openocd repo or in upstream openocd, but

gdb is sending:

```
vFlashErase:10000000,00007000
vFlashErase:10020000,00000000
```

0 size means erase all of flash which is invalid without the address starting at the beginning of flash. This probably has something to do with your memory segment starting at `10020000` being 1 byte big, and getting rounded down somewhere.

alexandruradovici, 2021-04-09:
> Yes, i just took a look, and I note that it doesn't look like your are using the pico-sdk, so this probably belongs in openocd, or gdb but.
> 
> gdb is sending:
> 
> ```
> vFlashErase:10000000,00007000
> vFlashErase:10020000,00000000
> ```
How did you see the issued commands? This is really useful.

> 
> 0 size means erase all of flash which is invalid without the address starting at the beginning of a flash bank. This probably has something to do with your memory segment starting at `10020000` being 1 byte big, and getting rounded down somewhere.

I will look into it, thank you.

kilograham, 2021-04-09:
I had a breakpoint in a sensible place in openocd :-) but you can probably get it by turning up the log level via the -d comand line option say `-d 2` or perhaps `-d 1`

alexandruradovici, 2021-04-10:
Thank you for the tips. I modified the section so that is has 8 bytes (0xff each) and attached the elf file. I get the following output with gdb:

```
(gdb) target remote :3333
Remote debugging using :3333
0x0000018a in ?? ()
(gdb) load
Loading section .text, size 0x100 lma 0x10000000
Loading section .text, size 0x7360 lma 0x10000100
Loading section .ARM.exidx, size 0x10 lma 0x10007460
Loading section .storage, size 0xb90 lma 0x10007470
Loading section .apps, size 0x8 lma 0x10020000
Error finishing flash operation
```

I have attached the openocd logs (logs2 with -d2, logs3 with -d3).
[raspberry_pi_pico.zip](https://github.com/raspberrypi/pico-sdk/files/6289719/raspberry_pi_pico.zip)