raspberrypi/pico-sdk #288: USB serial not working on hello_multicore example (race?)

yschaeff, 2021-03-28:
I ran the hello_multicore example with
```
pico_enable_stdio_usb(hello_multicore 1)
pico_enable_stdio_uart(hello_multicore 0)
```
in the _CmakeLists.txt_. But it did not present me with the expected _/dev/ttyACM0_. It also didn't mention anything in _dmesg_. After some experimentation I found out that it seems timing related. To reliably get a ttyACM0 I had to add a sleep at the end of the main function:
```
int main() {
    stdio_init_all();
// ... multicore specific code
    sleep_ms(600);
}
```
adding the sleep or a loop somewhere else in de main function also works.

lurch, 2021-03-28:
Sounds like [hello_multicore](https://github.com/raspberrypi/pico-examples/blob/master/multicore/hello_multicore/multicore.c) has probably finished executing `main` before the USB connection has had time to initialise, which is why you never see a `/dev/ttyACM0` device appearing. Switching from stdio_uart to stdio_usb is likely to present a similar problem with _any_ of the examples that don't explicitly keep running in a loop.

@aallan This should probably be moved to the pico-feedback repo and given a `documentation` label, and I'll try to add some explanatory words to section 4.1. of the getting-started guide.

yschaeff, 2021-03-28:
> I'll try to add some explanatory words to section 4.1. of the getting-started guide

Sounds great, I would have definitely picked that up if it where presented there.