raspberrypi/pico-sdk #292: Overhead in single SPI calls

marcelvanherk, 2021-03-30:
I found SPI slow on the pico.

This is a test program:

#include <hardware/spi.h>
#include <hardware/gpio.h>

spi_cpol_t cccpol;
spi_cpha_t cccpha;

void setup() {
gpio_set_function(16, GPIO_FUNC_SPI);
gpio_set_function(17, GPIO_FUNC_SPI);
gpio_set_function(18, GPIO_FUNC_SPI);
gpio_set_function(19, GPIO_FUNC_SPI);

spi_init(spi0, 100000000);
cccpol = SPI_CPOL_0;
cccpha = SPI_CPHA_0;
}

void loop() {
for (int i=0; i<65536; i++) {
uint16_t ret16;
uint16_t data16=0;
uint8_t ret8;
uint8_t data8=0;
spi_set_format(spi0, 8, cccpol, cccpha, SPI_MSB_FIRST);
spi_write_read_blocking(spi0, &data8, &ret8, 1);
spi_set_format(spi0, 16, cccpol, cccpha, SPI_MSB_FIRST);
spi_write16_read16_blocking(spi0, &data16, &ret16, 1);
spi_write16_read16_blocking(spi0, &data16, &ret16, 1);
}
}`

This is the SPI clock signal, i division is 1us.

[image]

There is almost a 1us overhead per call. Is this expected? Thanks, Marcel

Bodmer, 2021-03-31:
I noticed this too.

Examining the SDK code indicates that it is more suited to bulk transfers due to the way it handles the SPI buffers.

If you send a block of data it is much faster, at 62.5MHz the gap between 16 bit values is ~30ns:
[image]

SPI clock rate is 62.5MHz but 'scope is only 50MHz bandwidth, hense the "Sine" clock! However the delay is representative.

I have the Pico running with an ILI9341 using the Arduino IDE and Earle's excellent core, [see here](url).  Results are good when blocks are sent to the TFT:
```
Benchmark,                Time (microseconds)
Screen fill (5 times),    114981  (43.49 fps)
Text,                     31986
Lines,                    128851
Horiz/Vert Lines,         18031
Rectangles (outline),     10210
Rectangles (filled),      241497
Circles (filled),         70702
Circles (outline),        58318
Triangles (outline),      33224
Triangles (filled),       109793
Rounded rects (outline),  30585
Rounded rects (filled),   255167
Total = 1103345us
Total = 1.1033s
```

Not so good for individual pixels though. However the Pico has RAM enough for a full screen buffer... and DMA...

kilograham, 2021-03-31:
This is not unreasonable; you are making a method call each time (which for something non trivial that saves/restores general purpose registers costs you about 35-40 cycles). Add to that the fact that those methods happen to be in RAM there is a little more overhead. So that's about 50 cycles before any works has been done, so that's 1/3 of a us right there.

The moral of the story, is if you want to do things fast on Cortex M0+ (or in general) don't call methods in the inner loop!.

The API allows you to do bulk writes, or you can use DMA, or if you have some weird hybrid inner loop you'll have to roll it yourself (you can copy the code from the existing functions)

closed 2021-03-31 by kilograham

marcelvanherk, 2021-04-01:
Your answer is not very satisfying and only partially correct.

Interestingly, if I use this API to do a software SPI implementation I
achieve the same (!) overall speed, so the function call overhead cannot be
that great. You can also see that in the trace, two calls have 1 us gaps,
one call a 0.8 us gap.

Of course this test program simulates behaviour in a more complex program,
the same program on a same speed Teensy3.5 runs twice as fast as on the
RP2040.

kilograham, 2021-04-01:
The bigger gaps presumably are because of the set_formats.

Teensy3.5 has an ARM Cortex M4 which has a much more dense instruction set. Copy the `spi_writeXX_readXX_blocking` methods into your source, renaming and make them `static inline`, and I bet you'll go a bunch faster.

Still, this switching between bit lengths seems like a good task for PIO

marcelvanherk, 2021-04-01:
Thanks. I will give that a try. But rp2040 does outperform teensy3.5 on
integer operations.

marcelvanherk, 2021-04-01:
Code with inline static:

`#include "SPI.h"
#include <hardware/spi.h>
#include <hardware/gpio.h>

spi_cpol_t cccpol;
spi_cpha_t cccpha;

void setup() {
   gpio_set_function(16, GPIO_FUNC_SPI);
   gpio_set_function(17, GPIO_FUNC_SPI);
   gpio_set_function(18, GPIO_FUNC_SPI);
   gpio_set_function(19, GPIO_FUNC_SPI);
    
   spi_init(spi0, 100000000); 
   cccpol = SPI_CPOL_0;
   cccpha = SPI_CPHA_0; 
}

static inline int qspi_write_read_blocking(spi_inst_t *spi, const uint8_t *src, uint8_t *dst, size_t len) {
    invalid_params_if(SPI, 0 > (int)len);

    // Never have more transfers in flight than will fit into the RX FIFO,
    // else FIFO will overflow if this code is heavily interrupted.
    const size_t fifo_depth = 8;
    size_t rx_remaining = len, tx_remaining = len;

    while (rx_remaining || tx_remaining) {
        if (tx_remaining && spi_is_writable(spi) && rx_remaining - tx_remaining < fifo_depth) {
            spi_get_hw(spi)->dr = (uint32_t) *src++;
            --tx_remaining;
        }
        if (rx_remaining && spi_is_readable(spi)) {
            *dst++ = (uint8_t) spi_get_hw(spi)->dr;
            --rx_remaining;
        }
    }

    return (int)len;
}

static inline int qspi_write16_read16_blocking(spi_inst_t *spi, const uint16_t *src, uint16_t *dst, size_t len) {
    invalid_params_if(SPI, 0 > (int)len);
    // Never have more transfers in flight than will fit into the RX FIFO,
    // else FIFO will overflow if this code is heavily interrupted.
    const size_t fifo_depth = 8;
    size_t rx_remaining = len, tx_remaining = len;

    while (rx_remaining || tx_remaining) {
        if (tx_remaining && spi_is_writable(spi) && rx_remaining - tx_remaining < fifo_depth) {
            spi_get_hw(spi)->dr = (uint32_t) *src++;
            --tx_remaining;
        }
        if (rx_remaining && spi_is_readable(spi)) {
            *dst++ = (uint16_t) spi_get_hw(spi)->dr;
            --rx_remaining;
        }
    }

    return (int)len;
}

static inline void qspi_set_format(spi_inst_t *spi, uint data_bits, spi_cpol_t cpol, spi_cpha_t cpha, __unused spi_order_t order) {
    invalid_params_if(SPI, data_bits < 4 || data_bits > 16);
    // LSB-first not supported on PL022:
    invalid_params_if(SPI, order != SPI_MSB_FIRST);
    invalid_params_if(SPI, cpol != SPI_CPOL_0 && cpol != SPI_CPOL_1);
    invalid_params_if(SPI, cpha != SPI_CPHA_0 && cpha != SPI_CPHA_1);
    hw_write_masked(&spi_get_hw(spi)->cr0,
                    ((uint)(data_bits - 1)) << SPI_SSPCR0_DSS_LSB |
                    ((uint)cpol) << SPI_SSPCR0_SPO_LSB |
                    ((uint)cpha) << SPI_SSPCR0_SPH_LSB,
        SPI_SSPCR0_DSS_BITS |
        SPI_SSPCR0_SPO_BITS |
        SPI_SSPCR0_SPH_BITS);
}

void loop() {
  for (int i=0; i<65536; i++) {
    uint16_t ret16;
    uint16_t data16=0;
    uint8_t ret8;
    uint8_t data8=0;
    qspi_set_format(spi0, 8, cccpol, cccpha, SPI_MSB_FIRST);
    qspi_write_read_blocking(spi0, &data8, &ret8, 1);
    qspi_set_format(spi0, 16, cccpol, cccpha, SPI_MSB_FIRST);
    qspi_write16_read16_blocking(spi0, &data16, &ret16, 1);
    qspi_write16_read16_blocking(spi0, &data16, &ret16, 1);
  }
}`

marcelvanherk, 2021-04-01:
[image]

Trace: delays reduced to 400-600ns. Interesting, delay switching from 16 to 8 bits is longer than from 8 to 16 bits.

regards, Marcel

lurch, 2021-04-01:
P.S. To do multi-line code-blocks in github comments, you need 3 bacticks on a line by themselves, then all your lines of code, and then finally 3 bacticks on a line by themselves.

Bodmer, 2021-04-01:
I agree with @marcelvanherk, the instruction density argument does not explain the long delays.

There is something else going on here and it needs someone with a deep understanding of the way the underlying PL022 hardware works or has been implemented to explain it.

kilograham, 2021-04-01:
well the function overhead was clearly a problem... and the fasted the code could go inlined would be about 300ns. The remainder of the problem is likely that the max speed of the SPI is clksys/2 i.e. 62.5Mhz in this case

Bodmer, 2021-04-01:
No, @marcelvanherk re-wrote the code flow to optimise the use of the FIFO and avoid flow checks. Tests indicate function overhead is negligible compared to other delays.

Bodmer, 2021-04-01:
The main culprit seems to be this line:
```
spi_get_hw(spi0)->icr = SPI_SSPICR_RORIC_BITS;
```
This adds 300ns. Maybe the compiler is refusing to inline...

Bodmer, 2021-04-01:
Making some progress is mnimising delays between 8 and 16 bit transactions by writing a new dedicated function:
[image]

kilograham, 2021-04-01:
most of the time in the format functions is caused by the `cccpol`, `cccpha` parameters not being compile time constants (which they could be in this case)

kilograham, 2021-04-01:
> spi_get_hw(spi0)->icr = SPI_SSPICR_RORIC_BITS;

this isn't in the code above; if you are going to discuss your own code, then maybe post it.

kilograham, 2021-04-01:
note that 16 bits at 62.5Mhz is 256ns, so you are quite likely to wait up to that long for things to become not busy at various points in the code which sync with the hardware.

the code with the icr, should take a handful of cycles, unless you are explicitly running a debug build with all optimization turned off.

Bodmer, 2021-04-01:
There is nothing fundamentally wrong with the SDK code for the SPI writes, it is just that it is clear for single byte/word write only transfers it is inefficient. The call does not exit until all data is sent and receive buffer flushed (reasonable)  so the software downstream cannot make use of the FIFO buffer transmission time for write only situation, even for low SPI clock rates.

There is indeed a circa 170ns function call overhead but all internal calls are inlined OK, so it appears  >80% of the delay is simply register access and bit checking.

The good news is that once all this is known then it is possble to write optimised code for transmit only situations that elminates most of these delays.

marcelvanherk, 2021-04-02:
I wonder if there would be place for two inline single SPI write functions
i.e., like spi_writeOne8 and spi_writeOne16 that starts with waiting until
the FIFO is empty, sets the format and then writes, returning directly.
This would allow overlapping of transmission and logic execution without
too much complexity. A potential issue of course is setting of C/D which
would have to be done carefully.

lurch, 2021-04-02:
Or some kind of async / callback API? :man_shrugging:

xoblite, 2024-07-17:
While this issue is long since closed - but even then maybe a bit prematurely? - I'm curious about any further findings or follow-ups on this topic. @marcelvanherk , @Bodmer et al - any comments?

_(To very briefly summarize my own current situation, it essentially boils down to a SPI device requiring a CS flip after every transaction, which means that no more than <= 32 bits are written at a time, etc, and the overall performance/overhead I'm seeing is not quite what I would expect. I will try the static inline approach suggested above too shortly, but as the discussion above ended without final conclusion(s) I figured I should ask what if anything has played out since. Thanks in advance. )_

Polar-Axis, 2024-07-18:
I wrote my own code in the end which is specific for transmit only functionality. Essentially it does not wait for the FIFO to flush, but checks for room in the FIFO before new data is written.

Subsequently I implemented  PIO SPI which is the fastest transfer approach.

As stated above, the SDK is fine for general use.

Bodmer, 2024-07-19:
See optimised tx only code here:

https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L308