raspberrypi/pico-sdk #300: gpio_acknowledge_irq() doc too vague

blippy, 2021-04-02:
The SDK says, for `gpio_acknowledge_irq()`: "Acknowledge a GPIO interrupt." 

But what does that actually _mean_, why is that important? Does it mean that an interrupt won't fire again until it has been acknowledged, or does it just clear a flag that says the interrupt has fired? Or something else?

lurch, 2021-04-03:
It _does_ also [tell you](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__gpio.html#gaf59b91aef9af4b4fbd9cd1bb389a3e80) "**events**	Bitmask of events to clear. See gpio_set_irq_enabled for details."

I'm afraid at this stage large parts of the SDK-documentation still need to be cross-referenced with the [RP2040 datasheet](https://rptl.io/rp2040-datasheet) in order to make sense.

And of course the entire SDK is open-source, so if you _really_ want to know what something's doing, you can [just look](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_gpio/gpio.c#L136) :wink:  But again, you need to cross-reference this with section 2.19.3. of the datasheet.

daveythacher, 2021-10-29:
This should be inline.

closed 2022-06-30 by kilograham