raspberrypi/pico-sdk #301: adc.c.obj is always built in release or stripped

michael-ring, 2021-04-02:
I have built pico objects with latest release 1.1.1 to use for my freepascal bindings in both debug and release mode, so far in older builds the obj file for adc.c.obj debug was bigger than the release version, now both debug+release are same size.

Commands used to build:
```
rm -rf build
mkdir build
cd build
PICO_BOARD=${board} cmake -DCMAKE_BUILD_TYPE=Release ..
make -j 8 blink adc_console i2c_bus_scan bme280_spi pio_ws2812
```

for building debug I instead start cmake with:

`PICO_BOARD=${board} cmake -DCMAKE_BUILD_TYPE=Debug ..
`

all other objects I built have different size as expected...

lurch, 2021-04-03:
Did you forget to delete your build directory between builds? I just tried this myself (using different build-directories for debug and release), and I see:
(with output edited for brevity)
```bash
$ find build.release -name adc.c.obj -exec ls -l {} \;
-rw-rw-r-- 1 andrew andrew 892 Apr  3 12:40 build.release/adc/adc_console/...../hardware_adc/adc.c.obj
$ find build.debug -name adc.c.obj -exec ls -l {} \;
-rw-rw-r-- 1 andrew andrew 4496 Apr  3 12:40 build.debug/adc/adc_console/...../hardware_adc/adc.c.obj

$ find build.release -name spi.c.obj -exec ls -l {} \;
-rw-rw-r-- 1 andrew andrew 3020 Apr  3 12:40 build.release/spi/bme280_spi/...../hardware_spi/spi.c.obj
$ find build.debug -name spi.c.obj -exec ls -l {} \;
-rw-rw-r-- 1 andrew andrew 18640 Apr  3 12:40 build.debug/spi/bme280_spi/...../hardware_spi/spi.c.obj
```

michael-ring, 2021-04-03:
Nope, I delete everything in the build directory except my units directory where I stare the objects I need.

Here's the result of a full run:

```
-rw-r--r--  1 ring  staff    920  3 Apr 14:15 adc.c-debug.obj
-rw-r--r--  1 ring  staff    920  3 Apr 14:15 adc.c.obj
-rw-r--r--  1 ring  staff    888  3 Apr 14:15 boot2_adafruit_feather_rp2040.obj
-rw-r--r--  1 ring  staff  12396  3 Apr 14:15 claim.c-debug.obj
-rw-r--r--  1 ring  staff   2160  3 Apr 14:15 claim.c.obj
-rw-r--r--  1 ring  staff  24616  3 Apr 14:15 clocks.c-debug.obj
-rw-r--r--  1 ring  staff   5236  3 Apr 14:15 clocks.c.obj
-rw-r--r--  1 ring  staff  25488  3 Apr 14:15 gpio.c-debug.obj
-rw-r--r--  1 ring  staff   4968  3 Apr 14:15 gpio.c.obj
-rw-r--r--  1 ring  staff  30936  3 Apr 14:15 i2c.c-debug.obj
-rw-r--r--  1 ring  staff   4716  3 Apr 14:15 i2c.c.obj
-rw-r--r--  1 ring  staff  39264  3 Apr 14:15 irq.c-debug.obj
-rw-r--r--  1 ring  staff   4840  3 Apr 14:15 irq.c.obj
-rw-r--r--  1 ring  staff  45936  3 Apr 14:15 pio.c-debug.obj
-rw-r--r--  1 ring  staff   6548  3 Apr 14:15 pio.c.obj
-rw-r--r--  1 ring  staff   5836  3 Apr 14:15 platform.c-debug.obj
-rw-r--r--  1 ring  staff   1044  3 Apr 14:15 platform.c.obj
-rw-r--r--  1 ring  staff   8132  3 Apr 14:15 pll.c-debug.obj
-rw-r--r--  1 ring  staff   1144  3 Apr 14:15 pll.c.obj
-rw-r--r--  1 ring  staff   3236  3 Apr 14:15 spi.c-debug.obj
-rw-r--r--  1 ring  staff   3236  3 Apr 14:15 spi.c.obj
-rw-r--r--  1 ring  staff  87016  3 Apr 14:15 time.c-debug.obj
-rw-r--r--  1 ring  staff  10552  3 Apr 14:15 time.c.obj
-rw-r--r--  1 ring  staff  25844  3 Apr 14:15 timer.c-debug.obj
-rw-r--r--  1 ring  staff   4404  3 Apr 14:15 timer.c.obj
-rw-r--r--  1 ring  staff  15416  3 Apr 14:15 uart.c-debug.obj
-rw-r--r--  1 ring  staff   2232  3 Apr 14:15 uart.c.obj
-rw-r--r--  1 ring  staff  14116  3 Apr 14:15 watchdog.c-debug.obj
-rw-r--r--  1 ring  staff   2540  3 Apr 14:15 watchdog.c.obj
-rw-r--r--  1 ring  staff   5324  3 Apr 14:15 xosc.c-debug.obj
-rw-r--r--  1 ring  staff   1240  3 Apr 14:15 xosc.c.obj
```
as you can see for most objects things worked as expected.

[buildallpico.sh.txt](https://github.com/raspberrypi/pico-sdk/files/6252540/buildallpico.sh.txt)

lurch, 2021-04-03:
Your script seems to be deleting the build directory for each board, but it's not deleting the build directory when switching from release to debug?

lurch, 2021-04-03:
Ahaha, I've found your problem... when you do `find . -name "${object}.c.obj" | head -1` after build your debug objs, it's finding the already-copied `units/adc.c.obj` and simply copying **that** to `units/adc.c-debug.obj` :rofl: 

If I change your `cp $OBJFILE units/` line to `cp $OBJFILE units/${object}.c-release.obj` then it works as expected:
```
-rw-rw-r-- 1 andrew andrew  4504 Apr  3 13:54 adc.c-debug.obj
-rw-rw-r-- 1 andrew andrew   892 Apr  3 13:54 adc.c-release.obj
-rw-rw-r-- 1 andrew andrew   888 Apr  3 13:54 boot2_pico.obj
-rw-rw-r-- 1 andrew andrew 10164 Apr  3 13:54 claim.c-debug.obj
-rw-rw-r-- 1 andrew andrew  2088 Apr  3 13:54 claim.c-release.obj
-rw-rw-r-- 1 andrew andrew 21204 Apr  3 13:54 clocks.c-debug.obj
-rw-rw-r-- 1 andrew andrew  5196 Apr  3 13:54 clocks.c-release.obj
-rw-rw-r-- 1 andrew andrew 22788 Apr  3 13:54 gpio.c-debug.obj
-rw-rw-r-- 1 andrew andrew  4928 Apr  3 13:54 gpio.c-release.obj
-rw-rw-r-- 1 andrew andrew 26332 Apr  3 13:54 i2c.c-debug.obj
-rw-rw-r-- 1 andrew andrew  4640 Apr  3 13:54 i2c.c-release.obj
-rw-rw-r-- 1 andrew andrew 30360 Apr  3 13:54 irq.c-debug.obj
-rw-rw-r-- 1 andrew andrew  4820 Apr  3 13:54 irq.c-release.obj
-rw-rw-r-- 1 andrew andrew 36044 Apr  3 13:54 pio.c-debug.obj
-rw-rw-r-- 1 andrew andrew  6512 Apr  3 13:54 pio.c-release.obj
-rw-rw-r-- 1 andrew andrew  5132 Apr  3 13:54 platform.c-debug.obj
-rw-rw-r-- 1 andrew andrew  1016 Apr  3 13:54 platform.c-release.obj
-rw-rw-r-- 1 andrew andrew  7120 Apr  3 13:54 pll.c-debug.obj
-rw-rw-r-- 1 andrew andrew  1128 Apr  3 13:54 pll.c-release.obj
-rw-rw-r-- 1 andrew andrew 18668 Apr  3 13:54 spi.c-debug.obj
-rw-rw-r-- 1 andrew andrew  3020 Apr  3 13:54 spi.c-release.obj
-rw-rw-r-- 1 andrew andrew 69832 Apr  3 13:54 time.c-debug.obj
-rw-rw-r-- 1 andrew andrew 10324 Apr  3 13:54 time.c-release.obj
-rw-rw-r-- 1 andrew andrew 19100 Apr  3 13:54 timer.c-debug.obj
-rw-rw-r-- 1 andrew andrew  4348 Apr  3 13:54 timer.c-release.obj
-rw-rw-r-- 1 andrew andrew 12964 Apr  3 13:54 uart.c-debug.obj
-rw-rw-r-- 1 andrew andrew  2192 Apr  3 13:54 uart.c-release.obj
-rw-rw-r-- 1 andrew andrew 12696 Apr  3 13:54 watchdog.c-debug.obj
-rw-rw-r-- 1 andrew andrew  2484 Apr  3 13:54 watchdog.c-release.obj
-rw-rw-r-- 1 andrew andrew  4588 Apr  3 13:54 xosc.c-debug.obj
-rw-rw-r-- 1 andrew andrew  1212 Apr  3 13:54 xosc.c-release.obj
```

michael-ring, 2021-04-03:
Yes, you are correct. 
Sorry for wasting your time!

closed 2021-04-03 by michael-ring