raspberrypi/pico-sdk #303: When tinyusb upstream repo will be used in the sdk?

kehribar, 2021-04-03:
Hi All,

According to [1], official tinyusb repo is the way to go for latest features and for _less_ bugs.

But pico-sdk still uses https://github.com/raspberrypi/tinyusb even for the latest release issued yesterday. 

When this repo will start using https://github.com/hathach/tinyusb as a submodule?

Thanks.

[1] https://github.com/raspberrypi/tinyusb/issues/4#issuecomment-778849350

lurch, 2021-04-03:
I believe it's on the todo list, but there's no ETA yet. See also #161

luxedo, 2021-04-07:
I spent quite some time to understand why the default gamepad descriptor wasn't working  😆

kilograham, 2021-04-07:
As @lurch says #161 which is in 1.2.0 which is the next feature release

closed 2021-04-07 by kilograham