raspberrypi/pico-sdk #305: i2c: values for tlow lower than the minimum values specified in the I2C specification

fivdi, 2021-04-04:
Per table 10 on page 48 of the [I<sup>2</sup>C-bus specification and user manual](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) the minimum values for t<sub>LOW</sub> and t<sub>HIGH</sub> are as follows:

Symbol | Parameter | Standard-mode | Fast-mode | Fast-mode Plus | Unit
:---: | :---: | :---: | :---: | :---: | :---
t<sub>LOW</sub> | LOW period of the SCL clock | 4.7 | 1.3 | 0.5 | μs
t<sub>HIGH</sub> | HIGH period of the SCL clock | 4.0 | 0.6 | 0.26 | μs

With pico-sdk v1.1.0, the actual values used for t<sub>LOW</sub> are lower than the minimum values specified in the specification (<sup>1</sup>):

Symbol | Parameter | Standard-mode | Fast-mode | Fast-mode Plus | Unit
:---: | :---: | :---: | :---: | :---: | :---
t<sub>LOW</sub> | LOW period of the SCL clock | 4.0<sup>1</sup> | 1.008<sup>1</sup> | 0.4<sup>1</sup> | μs
t<sub>HIGH</sub> | HIGH period of the SCL clock | 6.0 | 1.496 | 0.6 | μs

The corresponding values written to the `IC_FS_SCL_LCNT` and `IC_FS_SCL_HCNT` registers are:

Symbol | Register | Standard-mode | Fast-mode | Fast-mode Plus
:---: | :---: | :---: | :---: | :---:
lcnt | IC_FS_SCL_LCNT | 500 | 126 | 50
hcnt | IC_FS_SCL_HCNT | 750 | 187 | 75

This isn't resulting in any issues with the I<sup>2</sup>C devices that I'm using with pico-sdk but it would probably be worth changing. Perhaps the values written to `lcnt` should have been written to `hcnt` and vice versa? See [here](https://github.com/raspberrypi/pico-sdk/blob/1.1.0/src/rp2_common/hardware_i2c/i2c.c#L69-L71).

fivdi, 2021-04-06:
I just noticed that @kevinjwalters also spotted this issue on a logic analyzer [here](https://github.com/raspberrypi/pico-sdk/issues/283#issuecomment-808746340).

kevinjwalters, 2021-04-07:
@fivdi Yes, I was hoping for someone else to confirm my analysis before I put a ticket in for it :)

Wren6991, 2021-04-09:
#314 is now merged, thank you @fivdi

PR #314 (2021-04-09): i2c: set high and low times to values that conform to the i2c specification
Fixes #305.
closed 2021-04-09 by Wren6991