raspberrypi/pico-sdk #308: Build fails with fatal error U1033

0xffset, 2021-04-06:
On windows following the getting started the compilation doesn't output any uf2 or hex files and fails at the second compilation with this error.
`Syntax Error: ":" unexpected in compiler_depend.make`
How can this be fixed if the guide has been followed to the exact word?

```[build] CMakeFiles\LED.dir\compiler_depend.make(4) : fatal error U1033: Syntaxfehler: ":" unerwartet
[build] Stop.
[build] NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29910\bin\HostX86\x86\nmake.exe"": R�ckgabe-Code "0x2"
[build] Stop.
[build] NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29910\bin\HostX86\x86\nmake.exe"": R�ckgabe-Code "0x2"
[build] Stop.```

lurch, 2021-04-06:
This is a problem with CMake 3.20.0 - it should be fixed in CMake 3.20.1, but for now you'll need to downgrade to CMake 3.19.x

0xffset, 2021-04-06:
wow, there are no words to show how thankful i am. Thanks.

closed 2021-04-06 by 0xffset