raspberrypi/pico-sdk #313: code/data in section ".scratch_x" is not included in data_cpy_table

bithoarder, 2021-04-06:
Code assigned to section ".scratch_x" is linked against address 0x20040000, but does not get copied from flash to ram at startup. 

I don't know linker scripts well, but my guess is that `*(.scratch_x.*)` should have been `*(.scratch_x*)`.

```
    .scratch_x : {
        __scratch_x_start__ = .;
        *(.scratch_x.*)
        . = ALIGN(4);
        __scratch_x_end__ = .;
    } > SCRATCH_X AT > FLASH
    __scratch_x_source__ = LOADADDR(.scratch_x);
```

Repo case:
edit the blink.c example to:
int __attribute__ ((section (".scratch_x"))) main() {

kilograham, 2021-04-08:
Ah most things use the `__scratch_x(foo)` macro which does `scratch_x.foo`

That it appears at all at `0x2004000`? is weird because of that... does it appear before `__scratch_x_start__`? i.e. is `__scratch_x_start__` pushed down? If so that is a lucky/unlucky `ld` feature i guess. might explain some weirdness from the mists of time ways back where my assembly based scratch_x code was disappearing. I thought I had chalked that up to `"ax"` on the section, but might have gone away when I added a `.foo` to the section name!!

Sadly right now, there are a bunch of projects that have cloned our linker scripts to modify them, so changing it now might just cause more confusion.

FYI. We do want to add some templated linker script generation in a future release, so that people can make minor tweaks without forking the whole link script.

Wren6991, 2021-04-09:
> edit the blink.c example to:
int attribute ((section (".scratch_x"))) main() {

A bit of extra context, the reason the macro uses unique section names of the form `.scratch_x.foobar` is so that the linker can garbage collect unused functions, which you really want since the scratch RAM sections are quite small. If all of your functions are in a section called `.scratch_x` then they all live or die together.

Most functions get a section suffix added automatically (because we build with `-ffunction-sections`) but this needs to be done via macro if you are explicitly moving a function to a different section.

kilograham, 2021-05-31:
note we are unlikely to fix this since you can simply add a "."

closed 2021-05-31 by kilograham