raspberrypi/pico-sdk #315: Unable to UART TXE and RXE bits

alexandruradovici, 2021-04-06:
While porting Tock (written in Rust, I am not able to use the SDK libraries) to RP2040, I am trying to enable the UART transmitter. I think I have followed all the steps,. but setting the UARTCR TXE and RXE bits seems to have no effect. 

Setting UARTEN works, but writing 1 to TXE and RXE keeps the bits low. Does anyone have any idea why this might happen?

I did set the baudrate to 115200 (the computed value is 115207).

fivdi, 2021-04-06:
In C, this works:
```c
uart_get_hw(uart)->cr = UART_UARTCR_UARTEN_BITS | UART_UARTCR_TXE_BITS | UART_UARTCR_RXE_BITS;
```
and this works
```c
uart_get_hw(uart)->cr = UART_UARTCR_TXE_BITS | UART_UARTCR_RXE_BITS;
uart_get_hw(uart)->cr |= UART_UARTCR_UARTEN_BITS;
```
but this doesn't:
```c
uart_get_hw(uart)->cr |= UART_UARTCR_UARTEN_BITS;
uart_get_hw(uart)->cr = UART_UARTCR_TXE_BITS | UART_UARTCR_RXE_BITS;
```
The order in which the bits are set is important. Make sure UARTEN is set at the same time as or after TXE and RXE.

fivdi, 2021-04-06:
Sorry, I now see that there was an error in the third example above. It should have been:
```c
    uart_get_hw(uart)->cr = UART_UARTCR_UARTEN_BITS ;
    uart_get_hw(uart)->cr |= UART_UARTCR_TXE_BITS | UART_UARTCR_RXE_BITS;
```
and, of course, it does work.

fivdi, 2021-04-06:
Is the baudrate being set before the UART is being enabled as is the case in the pico-sdk [here](https://github.com/raspberrypi/pico-sdk/blob/1.1.1/src/rp2_common/hardware_uart/uart.c#L52-L57)?

lurch, 2021-04-06:
The RP2040 uses PL011 UARTs.
https://developer.arm.com/documentation/ddi0183/g/programmers-model/register-descriptions/control-register--uartcr?lang=en says:

Program the control registers as follows:
1. Disable the UART.
1. Wait for the end of transmission or reception of the current character.
1. Flush the transmit FIFO by setting the FEN bit to 0 in the Line Control Register, UARTLCR_H.
1. Reprogram the UARTCR Register.
1. Enable the UART.

BTW @fivdi I believe the intention for the `*_BITS` definitions is to use them as bitmasks, rather than using them to directly set values :wink:

fivdi, 2021-04-06:
> BTW @fivdi I believe the intention for the `*_BITS` definitions is to use them as bitmasks, rather than using them to directly set values 😉 

Well, the pico-sdk uses the `*_BITS` definitions to directly set values quite a bit so that's where I'm getting my bad habits from :smile:

lurch, 2021-04-06:
Hmmm, perhaps I should have kept :zipper_mouth_face: