raspberrypi/pico-sdk #318: adc_read() for onboard temperature sensor return wrong value

Dracanrage, 2021-04-07:
The value of adc_read() for onboard temperature sensor returns wrong value, when the pico rebooted in the second time.(The first time of reboot is after flash the .uf2 file).
For instance, The temperature value is 26°C in the first time, after reboot (use RUN pin or other ways besides flash a new program), it will be 406°C。
Here is my code:
```
// reads the on board temparature in C 
int temperature(){
    // init ADC only if it has not been set up yet
    PinSetupADC::instance().initADC();
    // this is just setting some flags - so it does not hurt to call this redundantly
    adc_set_temp_sensor_enabled(true);

    //Input 4 is the onboard temperature sensor - operation only if there is a change
    PinSetupADC::instance().adcSelect(4);
    uint16_t value = 0;
    for (uint8_t i = 0; i < 30 ;i++){
        value = (value + adc_read())/2;
    }
    //T = 27 - (ADC_Voltage - 0.706)/0.001721
    //The ADC_Voltage is equal to adc_read() * 3.3 / (1<<12)
    const float conversion_factor = 3.3f / (1<<12);
    int temperature = 27 - (value * conversion_factor - 0.706)/0.001721;
    return temperature;
}
```

lurch, 2021-04-07:
`PinSetupADC::instance()` doesn't look like code written using `pico-sdk`, but some other framework?
Can you provide an (complete) example that shows the problem using only `pico-sdk` code please?

Dracanrage, 2021-04-07:
> `PinSetupADC::instance()` doesn't look like code written using `pico-sdk`, but some other framework?
> Can you provide an (complete) example that shows the problem using only `pico-sdk` code please?

I have found it was caused by the library which I used, now it was solved.
Thank you very much!

closed 2021-04-07 by Dracanrage