raspberrypi/pico-sdk #327: OpenOCD error: Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0

alexandruradovici, 2021-04-09:
I have a Raspberry Pi Pico connected to a Raspberry Pi 3.

I am getting a strange openocd error:

```
pi@raspberrypi:~ $ openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-02-23-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 1001 kHz
Info : Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0
Info : DAP init failed

Info : Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0
```

I tried booting with BOOTSEL pressed, same error.

fivdi, 2021-04-09:
This issue doesn't appear to be related to pico-sdk. I can reproduce this issue by disconnecting SWDIO so I would imagine that the Raspberry Pi and Pico are not correctly connected to each other.

```
pi@raspberrypi:~ $ openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-03-05-22:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 1001 kHz
Info : Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0
Info : DAP init failed

Info : Read incorrect DLIPDR 00000000 (possibly CTRL/STAT value) when selecting coreid 0
pi@raspberrypi:~ $ 
```

alexandruradovici, 2021-04-09:
You are right, it was a powering problem.

closed 2021-04-09 by alexandruradovici

hristijandailovski, 2023-07-28:
I have the same problem. I power my pico by a 5v pin from the Rp pi 3 a+ board. The jumper wire is connected to VSYS pin on pico. I use copper wires to connect  SWDIO ,SWCLK, and GND. GPIO 25 and GPIO 24, are used for SWCLK and SWDIO, respectively. Is there solution to the problem ?

jasvinder1079, 2023-10-09:
getting same issue even i cross check my power connections. Any suggestion?

lurch, 2023-10-14:
Are you following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?
For questions like these you should ask for help at https://forums.raspberrypi.com/viewforum.php?f=143 rather than on GitHub.