raspberrypi/pico-sdk #328: missing defines default SPI pins for sparkfun boards

2bndy5, 2021-04-09:
I noticed that [the header files](https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards) for the sparkfun boards are missing some `#define`s for the default SPI pins. However, these boards have specifically broken out SPI pins to the boards' pins (labeled as SPI pins on the board).

Here's a list of default SPI pins that my research has turned up (with links to schematics):
### [Sparkfun Thing Plus](https://cdn.sparkfun.com/assets/5/4/f/6/b/RP2040_Thing_Plus_Schematic.pdf)
```cpp
#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_SCK_PIN 2
#endif

#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_TX_PIN 3
#endif

#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_RX_PIN 4
#endif
```
### [Sparkfun Pro Micro](https://cdn.sparkfun.com/assets/4/a/6/d/6/SparkFun_Pro_Micro_RP2040_Schematic.pdf) & [Sparkfun MicroMod](https://cdn.sparkfun.com/assets/1/8/3/5/8/MicroMod-RP2040-ProcessorBoard_Schematic.pdf)
```cpp
#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_SCK_PIN 22
#endif

#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_TX_PIN 23
#endif

#ifndef PICO_DEFAULT_SPI_SCK_PIN
#define PICO_DEFAULT_SPI_RX_PIN 20
#endif
```

I also looked into the pimoroni boards, but there's discrepencies:
- [pimoroni_picosystem](https://shop.pimoroni.com/products/picosystem) store page is missing links to schematics/pinout (though it does boast that SPI is exposed)
- [pimoroni_tiny2040](https://shop.pimoroni.com/products/tiny-2040) boards' pins are not specifically labeled for SPI usage (though multiple options exist to use `spi0`)
- [pimoroni_keybow2040](https://shop.pimoroni.com/products/keybow-2040?variant=32399559589971) doesn't seem to provide accessible GPIO pins (or more importantly no SPI bus exposed)

lurch, 2021-04-10:
> I noticed that the header files for the sparkfun boards are missing some #defines for the default SPI pins.

ping @kirk-sfe @nseidle 

> I also looked into the pimoroni boards, but there's discrepencies:

ping @ZodiusInfuser @Gadgetoid

ZodiusInfuser, 2021-04-16:
> * [pimoroni_picosystem](https://shop.pimoroni.com/products/picosystem) store page is missing links to schematics/pinout (though it does boast that SPI is exposed)

That section of the store page is mentioning the capabilities of the RP2040, rather than the PicoSystem's IO, so the already provided pinout is accurate.

> * [pimoroni_tiny2040](https://shop.pimoroni.com/products/tiny-2040) boards' pins are not specifically labeled for SPI usage (though multiple options exist to use `spi0`). So, the pimoroni_tiny2040.h could benefit from
>   ```c++
>   #ifndef PICO_DEFAULT_SPI
>   #define PICO_DEFAULT_SPI 0
>   #endif
>   ```

Yes, we don't specify a default as it is up to the user which they use. You make a good point about SPI0 only being accessible though so perhaps your suggestion would be a good addition.

> * [pimoroni_keybow2040](https://shop.pimoroni.com/products/keybow-2040?variant=32399559589971) doesn't seem to provide accessible GPIO pins (or more importantly no SPI bus exposed)

Some GPIO is exposed on Keybow 2040, but not much as most is reserved for internal functions. You can see the exposed pins in this Tom's hardware picture: https://cdn.mos.cms.futurecdn.net/PiaJPCoQYqLh3pGfShNqSA-970-80.jpg.webp

ZodiusInfuser, 2021-04-16:
@lurch I notice things have been updated since I added the previous Pim board definitions, so let me know if there's anything I should update, either define wise, or styling wise.

Also, am I correct in remembering that the system will default to defines from pico.h if we do not provide them? I'm in the process of creating a PR for a new board 😉

2bndy5, 2021-04-16:
In my experience, the system only defaults to pico.h if `PICO_BOARD` is not defined (I think the SDK doc says this also). Although, I think one could bring in the defaults from _pico.h_ if they include the file at the end of the header file for a different board (there's a [comment about this tactic in the _pico.h_ file](https://github.com/raspberrypi/pico-sdk/blob/afc10f3599c27147a6f34781b7102d86f58aa5f6/src/boards/include/boards/pico.h#L12)).

ZodiusInfuser, 2021-04-16:
PR made with updated pimoroni board definitions #343 (and a teaser), if you wish to check over it @2bndy5.

2bndy5, 2021-04-16:
@ZodiusInfuser I found the changes to default SPI pins on the picosystem most teasing 😮 . 

### Concerning the tiny2040
>we don't specify a default as it is up to the user which they use

I'm ok with keeping this open-ended (not defining the default SPI pins) as long as the board's silkscreen doesn't hint the user to choose a specific option. Though, the `PICO_DEFAULT_SPI` does seem helpful for people not used to an identifying number for which SPI bus to use. In fact, defining `PICO_DEFAULT_SPI` should aid in the selection process from [the docs GPIO Function Select Table](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__gpio.html#details)

ZodiusInfuser, 2021-04-16:
> @ZodiusInfuser I found the changes to default SPI pins on the picosystem most teasing 😮 .

That wasn't the tease I was referring to 😉 
 
> ### Concerning the tiny2040
>
> I'm ok with keeping this open-ended (not defining the default SPI pins) as long as the board's silkscreen doesn't hint the user to choose a specific option. Though, the [...]

Yea, I thought about this after I posted and decided having some defaults would be useful, so have set some up for both I2C and SPI in the previous pull request.

2bndy5, 2021-04-16:
>That wasn't the tease I was referring to 😉

I'm guessing from the file names, you folks got QSPI working for various sized FLASH storage 🚀 I remember Lady Ada saying that she quickly moved past this issue when the 8MB flash chip seemed to work better than her usual choice of 2MB chips.

lurch, 2021-04-16:
> QSPI working for various sized FLASH storage

There was some work done by @kilograham in `pico-sdk` recently to make it easier to change which Flash chip is used-by-default for a particular board.

kirk-sfe, 2021-04-16:
Just added the the SPI definitions for the SparkFun boards. PR: #344

kilograham, 2021-04-18:
fixed by #343 and #344 (I think :-) )

closed 2021-04-18 by kilograham

2bndy5, 2021-04-18:
closing this as PRs #344 & #343 are merged into the develop branch 🎉

2bndy5, 2021-05-21:
@kirk-sfe I noticed the board headers for the ThingPlus, MicroMod, & ProMicro don't have an identifying macro like 
```cpp
// For board detection
#define SPARKFUN_THINGPLUS
```

All the other board headers have a macro to identify the board specified by `-DPICO_BOARD=sparkfun_thingplus` except for the RPi Pico board (but that's the assumed default board if `PICO_BOARD` isn't specified).

I'm mostly concerned about the ThingPlus because it has a rather unique wiring schematic compared to other SparkFun boards.