raspberrypi/pico-sdk #330: Today download install Errors

K2AAE, 2021-04-10:
New PI 4+8GB /128GB USB - Clean with LAMP installed. 4/10/2021 Python 3.9.4
From https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf followed the steps to the point of trying to enter hello_usb but there is a menu difference, I had no drop down list. I Clicked around and ended up building all, but:

-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built
I checked the 'pico/pico_examples/build/hello_world/usb/hello_usb.uf2 was there with 65.5KiB

lurch, 2021-04-10:
> -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built

Those are only warnings, they're safe to ignore.

> I checked the 'pico/pico_examples/build/hello_world/usb/hello_usb.uf2 was there with 65.5KiB

Yup, that looks perfectly fine.

When talking about "drop down list" are you referring to VS Code? There was a late-change where we split the single hello_world example up into hello_serial and hello_usb so it's possible some of the VS Code instructions don't match up. I'll check next week.
BTW I'm not sure what the LAMP stack has to do with anything Pico-related?

@kilograham Please move this to the pico-feedback repo.

kilograham, 2021-04-15:
Note i have change certain info about things that will need to be done to DEBUG level output also, so there is less spam in the build.