raspberrypi/pico-sdk #332: GPIO interrupt

hesolium, 2021-04-11:
Hi, 
I'm trying to migrate radio RFM23B library to RPI Pico. The radio is connected via SPI and after connecting the cables I can communicate with it. But I can't use interrupts to handle events. The interrupt handler is called correctly and handles the radio (e.g. receiving a packet), but just after handling the interrupt, the program is terminated with an isr_hardfault event (by core number 1). My interrupt routine is relatively long. It has to read (via SPI) the radio status and handle events (eg copying via SPI the packet to my buffer). I try to block all interrupts while the handler is running. But the error shows up after unlocking interrupts. My Pico is connected with Picoprobe (for debugging). How to find the cause? Any ideas?

kilograham, 2021-04-11:
I suggest you ask questions on the forums here: https://www.raspberrypi.org/forums/viewforum.php?f=144 - do you actually mean core 1? if so what do you mean by "I try to block all interrupts"? Are you running an RTOS; what is the second core doing etc... some more details would help, but as I say ask over there

closed 2021-04-11 by kilograham

hesolium, 2021-04-11:
@kilograham:"do you actually mean core 1?"
Sorry for the inaccuracy. Not core 1 but Thread 1. I don't use RTOS. The application uses pure Pico C-SDK (bare-metal).
 I suspect a bug in the SDK because my function is just a callback. Interrupt handling is performed by the static SDK function (gpio_irq_handler). What isr_hardfault mean? Has anything to do with interrupts? I can't find any meaningful description.

kilograham, 2021-04-11:
isr_hardfault is the exception handler for a hard fault. just google hard fault. It is likely either a bug in your callback, or you passed a bad funcrtion pointer as the callback.

kilograham, 2021-04-11:
This is quite involved but helpful https://interrupt.memfault.com/blog/cortex-m-fault-debug

hesolium, 2021-04-11:
@kilograham:"It is likely either a bug in your callback, or you passed a bad funcrtion pointer as the callback."
The first is possible :) The second is not. The callback function ran at least once (under the dubugger) and received one packet. Is it possible to install my own ISR in place of the standard 'isr_hardfault'?

kilograham, 2021-04-11:
simplest way is just to define your own function with the same name; doesn't probably help much... better to look at the stack as described in the link above to see where the fault occurred

hesolium, 2021-04-11:
Thank you for the link. I will try to analyze the situation. 
I usually work on the MBED and NRF52 platform. Maybe I can diagnose the problem myself.
Thanks.