raspberrypi/pico-sdk #333: USB serial doesn't open immediately after reset

andrewjherbert, 2021-04-11:
If my program commences output to the USBserial port immediately after reset the initial output is lost.  

Putting a 1000ms delay at the start of my program cures the problem.

Another work-around is as follows:

#include "tusb.h"
int main() {
    stdio_init_all();
    adc_init();
    printf("waiting for usb host");
    while (!tud_cdc_connected()) {
      printf(".");
      sleep_ms(500);
    }
    printf("\nusb host detected!\n");

But this too appears to be time sensitive - if I make the sleep <= 10ms I still lose the first line.

It would be nice to have a helper function that gets this right, or an option on opening the port.

Thanks.

lurch, 2021-04-11:
Related to #288 and https://github.com/raspberrypi/pico-examples/pull/60

I know very little about the lowlevels of USB, but I think some of the timing-sensitive delay might be partly due to the USB enumeration-process happening on the host?

andrewjherbert, 2021-04-12:
> Related to #288 <https://github.com/raspberrypi/pico-sdk/issues/288> and raspberrypi/pico-examples#60 <https://github.com/raspberrypi/pico-examples/pull/60>

Good thought, and I guess this will arise every time the Pico restarts as the host will detect the disconnect / reconnect?

The host is a RPi4 running Raspbian.

I guess the solution is to use UART serial but that involves another set of cables and in my application I have mortgaged most of the GPIO pins to other purposes.  

Am I correct in understanding that I can connect to any one set of UART GPIO pins for this purpose, I don’t have to use the first set as per the examples, or do I have to do something to point the Pico at the right set?

Cheers,

Andrew (aka EDSAC)

lurch, 2021-04-12:
> Am I correct in understanding that I can connect to any one set of UART GPIO pins for this purpose, I don’t have to use the first set as per the examples, or do I have to do something to point the Pico at the right set?

There's some handy defines in https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/pico.h :smiley: 
To override their values, just define them in your own application code before including any pico-sdk headers.

andrewjherbert, 2021-04-12:
Great, thanks!

A.

lurch, 2021-04-12:
From one Andrew to another, you're welcome! :grin:

kilograham, 2021-05-31:
having a "wait for connected" method should be a separate issue (there may be one)

closed 2021-05-31 by kilograham