raspberrypi/pico-sdk #342: multicore_fifo_wready() SDK definition inverted

ptj5210, 2021-04-15:
The SDK says multicore_fifo_wready() is "true if the FIFO is full, false otherwise", however the underlying register bit (FIFO_ST Register: RDY) says the "Value is 1 if this core’s TX FIFO is not full (i.e. if FIFO_WR is ready for more data)".   Seems that multicore_fifo_wready() is actually true if the FIFO is _not_ full.

lurch, 2021-04-15:
I think this is a duplicate of #235 ? That was already fixed by #239 

(although the fix is currently only on the [develop](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_multicore/include/pico/multicore.h#L90) branch and not yet on the [master](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_multicore/include/pico/multicore.h#L90) branch)

@kilograham Does some kind of "milestone" or something need to be added to that PR, to ensure that the change eventually works its way into `master` ? Or will that automatically happen during the next non-patch SDK release?

ptj5210, 2021-04-16:
Yep, it's a duplicate.  Sorry about that.

closed 2021-04-16 by ptj5210