raspberrypi/pico-sdk #349: Add comment pointing to documentation

debuti, 2021-04-20:
I arrived at  https://github.com/raspberrypi/pico-sdk/blob/afc10f3599c27147a6f34781b7102d86f58aa5f6/src/rp2_common/pico_multicore/multicore.c#L130:

`   uint32_t cmd_sequence[] = {0, 0, 1, (uintptr_t) vector_table, (uintptr_t) sp, (uintptr_t) entry};`

I miss here a link to the document where this command sequence is stated

lurch, 2021-04-20:
That might be the "launch protocol" referred to on page 156 of the RP2040 datasheet? So you might need to look at the bootrom sourcecode for more information? :man_shrugging:

debuti, 2021-04-20:
I did, and I found some routines in an assembly file. But Id like some solid docs so that I can port this to a Rust lib.

Thanks for the hint though

kilograham, 2021-04-28:
yes this is currently not documented, though as you say it would be useful to be... that said, if you re-implement what the C source does, then you are in good shape (namely that if core 1 is just coming out of reset - and you may want to if it was doing something else)... you will not suffer a race condition with the startup of core 1

kilograham, 2021-04-28:
p.s. that is the only meaningful sequence