raspberrypi/pico-sdk #350: Broken serial output

alexandruradovici, 2021-04-20:
I am using a Pico connected with three cables to serial0 of a Raspberry Pi. I tried using several types of Raspberry Pis (2, 3+, and 4), several cables and several Pico Pis. The output from the serial is mostly broken, due to several missing characters. This happens randomly, I am not able to find a rule.

Sometimes resetting the Pi Pico helps.

I tried running the hello_world over serial example.

I was wondering if anybody else had a similar issue?

fivdi, 2021-04-20:
I haven't experienced this problem and it works flawlessly for me when using a Pi 4 and a Pico.
Can you post a photo which clearly shows the Pico and how it is connected to the Pi?

fivdi, 2021-04-20:
Also, which program is being used on the Pi to view the Pico serial output.

lurch, 2021-04-20:
There was a report elsewhere (which I meant to try reproducing but haven't got round to, whoops) that claimed that using Bluetooth might be messing up the serial port?
https://github.com/raspberrypi/pico-feedback/issues/127

Can you double-check the serial-port settings on your Pi?

alexandruradovici, 2021-04-20:
> There was a report elsewhere (which I meant to try reproducing but haven't got round to, whoops) that claimed that using Bluetooth might be messing up the serial port?
> [raspberrypi/pico-feedback#127](https://github.com/raspberrypi/pico-feedback/issues/127)
> 
I'll add the dtb that disables the BLE and try again, thank you.
> Can you double-check the serial-port settings on your Pi? Did you use the `pico_setup.sh` script?

 No, I did not know it existed, I used raspi-config to setup the serial port.

alexandruradovici, 2021-04-20:
> Also, which program is being used on the Pi to view the Pico serial output.

I tried screen and minicom, same result.

fivdi, 2021-04-20:
If Bluetooth is enabled it could help to explain the problem on a Pi 3 or Pi 4 but not on a Pi 2.

When using raspi-config to configure serial, it's also important to answer the following question with "No":

```
Would you like a login shell to be accessible over serial?
```
If this question is answered with "Yes", there will be a login shell associated with the UART and it will read some of the serial input that is coming from the Pico.