raspberrypi/pico-sdk #352: Access tags are missing on register level in the SVD file

atanasuzunov, 2021-04-21:
According CMSIS SVD specification, the <access> tag is mandatory on register level:

_"If the elements \<size>, \<access>, \<resetValue>, and \<resetMask> have not been specified on a higher level, then these elements are mandatory on register level."_

That causes side effects with some IDEs.
If a register contains mixed read-write and read-only fields, a proper implementation could be: defining the register as read-write, and then overriding the access type with a read-only <access> tag on the field level.

lurch, 2021-04-21:
I was initially confused by this report of "access tags being missing", so I checked and they _are_ present on the field-level, just not at any higher-levels.

@atanasuzunov I guess as a temporary workaround (until we fix this) you could edit your SVD file to define a default read-write `access` tag at the [device](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_device.html) level?

@kilograham Assign this to me please.

Hmmm, looks like we're missing `resetMask` tags too, perhaps that was added to the SVD spec after I wrote the SVD-creation tool? Or maybe I just never noticed the sentence saying "these elements are mandatory"? :shrug: 
BTW I'm fairly sure the SVD file _did_ pass the [SVDConv](https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_SVDConv_pg.html) checks when I wrote the SVD-creation tool.

JustAnother1, 2022-11-10:
There is a validated SVD file for the RP2040 available for [download](https://chipselect.org/Raspberry%20Pi--RP2040.svd)

If there is still an issue with the file you can contact me directly [info@chipselect.org](mailto:info@chipselect.org)

File also attached:
[RP2040.svd.zip](https://github.com/raspberrypi/pico-sdk/files/9977039/RP2040.svd.zip) (compressed as github does not accept *.svd as attachment)

kilograham, 2023-05-26:
is this fixed?

lurch, 2023-06-01:
Nope, I don't think so. And it should probably be reassigned to @andygpz11 as I believe he's looking after the SVD-generation code now?

lurch, 2024-06-24:
Reassigning to @matiasilva as I think he might have been looking at the SVD stuff most recently?

kilograham, 2024-08-09:
SVDs were improved on SDK2.0.0 please re-test

closed 2024-08-09 by kilograham