raspberrypi/pico-sdk #353: Cannot open `stdlib.h` on WSL 2 with VSCode/CMake

icorbrey, 2021-04-22:
I've set up my project using [this tutorial](https://paulbupejr.com/raspberry-pi-pico-windows-development/), and everything's gone pretty smoothly. However, once everything was set up I opened up `blink.c` and got the following error when trying to include `pico/stdlib.h`:

```
cannot open source file "pico/version.h" (dependency of "pico/stdlib.h")
```

I haven't been able to find any advice on this issue other than [this post](https://www.raspberrypi.org/forums/viewtopic.php?t=299887) on the Raspberry Pi forums saying I need to change CMake's build generator. Doing so had no visible effect. Any advice?

lurch, 2021-04-22:
Is that a compiler error, or an intellisense / auto-complete error? If it's a compiler error then it sounds like you've not got the `PICO_SDK_PATH` environment variable pointing at the right place? If it's an intellisense error then you'll have to fiddle with the VSCode settings :wink: 

AFAIK we only officially support the Windows setup detailed in our [Getting Started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) so if you're using WSL you may have to rely on 3rd-party support?

@aallan This should probably be moved to pico-feedback?

kilograham, 2021-04-22:
Assuming this is intellisense and not an actual build error, it may just be because you have not built yet... that file is generated by the build.

icorbrey, 2021-04-22:
My CMake version was old (3.10.x) so I went ahead and upgraded to latest. However, from what I can tell CMake can't actually see my `PICO_SDK_PATH`. I originally set it in VSCode, then when that didn't work set it in both `~/.zshrc` and `~/.bashrc` to no avail.