raspberrypi/pico-sdk #354: fatal error: pico/version.h: No such file or directory

PaulskPt, 2021-04-23:
Error in file pico-sdk/src/common/pico_base/include/pico.h: line 17: '"#include "pico/version.h"
is caused by the fact the the name of the file in subfolder: 'pico-sdk/src/common/pico_base/include/pico/' is not 'version.h' but 'version.h.in'.

lurch, 2021-04-23:
This file is actually autogenerated by the build system. See also https://github.com/raspberrypi/pico-feedback/issues/148

kilograham, 2021-04-28:
you don't say what you are doing to get the error, sounds like duplicate of linked issue anyway (which is that you haven't built yet)

closed 2021-04-28 by kilograham

PaulskPt, 2021-04-28:
Problem solved. Re-installed the pico-sdk; checked installation of other parts of toolchain. Updated pimoroni's libraries for equipment used. Now the toolchain compiles and builds OK.  Thank you.