raspberrypi/pico-sdk #359: C SDK documentation: PIO JMP X-- clarification

snoring-pigeon, 2021-04-25:
Suggestion for clarification in datsheet (p.341) and C SDK (p. 59) docs:

JMP X11 is described as "scratch X non-zero, post-decrement"

I find the use of 'post' confusing - it could mean
 1) test if X is non-zero after/post decrementing it (reading as plain English)
 2) test if X is non-zero, decrementing it afterwards/post (reading as C programmer)

I think this could be clearer, e.g.

"scratch X non-xero, decrement after test if X > 0"

(assuming the decrement only happens if X > 0, rather than wrapping to 0xffffffff)

Reneg973, 2021-04-25:
If I read from left to right, I first see the red text 'X--'. Isn't that unambiguous?

snoring-pigeon, 2021-04-25:
Yes, "X--" is fine, coming from a C background.

It's the following phrasing - that x is "non-zero, post-decrement" that threw me. I find it easy to read that as "non-zero, after decrement", rather than "non-zero, with a post-decrement operation"

lurch, 2021-04-26:
Page 342 of the datasheet clarifies:
"JMP X-- and JMP Y-- always decrement scratch register X or Y, respectively. The decrement is not conditional on the
current value of the scratch register. The branch is conditioned on the initial value of the register, i.e. before the
decrement took place: if the register is initially nonzero, the branch is taken."

...which I _think_ answers your question?

snoring-pigeon, 2021-04-26:
Yes - thank you. I was looking at an older version of the datasheet that didn't have that (sorry!)

I assume a decrement from X==0 wraps around to 0xffffffff?