raspberrypi/pico-sdk #361: Usb: Host: Setup respecting max packet length

ismell, 2021-04-26:
When using the USB host, I noticed an enumeration problem. It looks like the max packet size is being ignored. Here is a lot showing the following:
1) Connect
2) Send `GET_DESCRIPTOR` with max length of 8.
2) Rx descriptor with max packet length of 8
3) Send `SET_ADDRESS`
4) Ack `SET_ADDRESS`
5) Send `GET_DESCRIPTOR` with max length of 18 (full descriptor)
6) Write buffer control: `buffer control (0x50100080) <- 0x6412` <- The buffer length should be 0x08, not 0x12.
7) Rx only 8 bytes and get `Short rx transfer`. This results in a corrupt device descriptor.
8) TinyUSB continues enumeration even when the device says there is 0 configuration descriptors. I'll file a different bug for this.

```
hcd_init
USBH DEVICE ATTACH
hcd_port_connect_status
hcd_port_speed_get
full speed
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Set dev addr 0 ep 0 = 0
Get 8 byte of Device Descriptor
Control Setup: R: DEVICE, T: STANDARD, D: IN, R: GET_DESCRIPTOR, V: DEVICE, DescIdx: 0, Idx: 0, Len: 8
80 06 00 01 00 00 08 00
hcd_setup_send dev_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hcd_port_speed_get
full speed
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 8
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 in total_len 8, start=1
Start transfer of total len 8 on ep 0 in
Last buf (8 bytes left)
buffer control (0x50100080) <- 0x6408
hcd_port_speed_get
full speed
buf_status 0x00000001
rx 8 bytes (buf_ctrl 0x0000e008)
Completed transfer of 8 bytes on ep 0 in
on EP 80 with 8 bytes
Control data:
  0000:  12 01 00 02 FF FF FF 08                          |........|
Len: 8/18, Type: DEVICE, USB: 0x200, Class: 0xff, SubClass: 0xff, Protocol: 0xff, PacketSize: 8,
hcd_edpt_xfer dev_addr 0, ep_addr 0x0, len 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 out total_len 0, start=1
Start transfer of total len 0 on ep 0 out
Last buf (0 bytes left)
buffer control (0x50100080) <- 0xe400
hcd_port_speed_get
full speed
buf_status 0x00000001
tx 0 bytes (buf_ctrl 0x00006000)
Completed transfer of 0 bytes on ep 0 out
on EP 00 with 0 bytes
Port reset
hcd_port_reset
Set Address
Control Setup: R: DEVICE, T: STANDARD, D: OUT, R: SET_ADDRESS, Val: 0x1, Idx: 0, Len: 0
00 05 01 00 00 00 00 00
hcd_setup_send dev_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hcd_port_speed_get
full speed
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 0
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 in total_len 0, start=1
Start transfer of total len 0 on ep 0 in
Last buf (0 bytes left)
buffer control (0x50100080) <- 0x6400
hcd_port_speed_get
full speed
buf_status 0x00000001
rx 0 bytes (buf_ctrl 0x0000e000)
Completed transfer of 0 bytes on ep 0 in
on EP 80 with 0 bytes
hcd_device_close 0
hcd_edpt_open dev_addr 1, ep_addr 0
hw_endpoint_init dev 1 ep 0 out xfer 0
dev 1 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Set dev addr 1 ep 0 = 0
Control Setup: R: DEVICE, T: STANDARD, D: IN, R: GET_DESCRIPTOR, V: DEVICE, DescIdx: 0, Idx: 0, Len: 18
80 06 00 01 00 00 12 00
hcd_setup_send dev_addr 1
hw_endpoint_init dev 1 ep 0 out xfer 0
dev 1 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hcd_port_speed_get
full speed
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 18
hw_endpoint_init dev 1 ep 0 in xfer 0
dev 1 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 in total_len 18, start=1
Start transfer of total len 18 on ep 0 in
Last buf (18 bytes left)
buffer control (0x50100080) <- 0x6412
hcd_port_speed_get
full speed
buf_status 0x00000001
rx 8 bytes (buf_ctrl 0x0000e008)
Short rx transfer
Completed transfer of 8 bytes on ep 0 in
on EP 80 with 8 bytes
Control data:
  0000:  12 01 00 02 FF FF FF 08 00 00 00 00 00 00 00 00  |................|
  0010:  00 00                                            |..|
Len: 18/18, Type: DEVICE, USB: 0x200, Class: 0xff, SubClass: 0xff, Protocol: 0xff, PacketSize: 8, VID: 0, PID: 0, Device: 0,
```

pico-sdk: `afc10f3 (HEAD -> master, tag: 1.1.2, origin/master, origin/HEAD) 1.1.2 release`

ismell, 2021-04-26:
I think the following might fix it:

```
diff --git a/src/portable/raspberrypi/rp2040/rp2040_usb.c b/src/portable/raspberrypi/rp2040/rp2040_usb.c
index a44607e9..4de205bb 100644
--- a/src/portable/raspberrypi/rp2040/rp2040_usb.c
+++ b/src/portable/raspberrypi/rp2040/rp2040_usb.c
@@ -151,7 +151,7 @@ void _hw_endpoint_xfer_start(struct hw_endpoint *ep, uint8_t *buffer, uint16_t t
     ep->total_len = total_len;
     ep->len = 0;
     // FIXME: What if low speed
-    ep->transfer_size = total_len > 64 ? 64 : total_len;
+    ep->transfer_size = total_len > ep->wMaxPacketSize ? ep->wMaxPacketSize : total_len;
     ep->active = true;
     ep->user_buf = buffer;
     // Recalculate if this is the last buffer
```

ismell, 2021-04-26:
I do get a different error now:

```
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 18
hw_endpoint_init dev 1 ep 0 in xfer 0 max 8
dev 1 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 in total_len 18, start=1
Start transfer of total len 18 on ep 0 in, max: 8
buffer control (0x50100080) <- 0x2408
hcd_port_speed_get
full speed
buf_status 0x00000001
rx 8 bytes (buf_ctrl 0x0000a008)
Last buf (10 bytes left)
buffer control (0x50100080) <- 0x440a
buf_status 0x00000001
rx 10 bytes (buf_ctrl 0x0000c008)
Completed transfer of 18 bytes on ep 0 in
on EP 80 with 18 bytes
Control data:
  0000:  12 01 00 02 FF FF FF 08 5E 04 8E 02 10 01 01 02  |........^.......|
  0010:  00 00                                            |..|
Len: 18/18, Type: DEVICE, USB: 0x200, Class: 0xff, SubClass: 0xff, Protocol: 0xff, PacketSize: 8, VID: 0x45e, PID: 0x28e, De
hcd_edpt_xfer dev_addr 1, ep_addr 0x0, len 0
hw_endpoint_init dev 1 ep 0 out xfer 0 max 8
dev 1 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
hw_endpoint_xfer ep 0 out total_len 0, start=1
Start transfer of total len 0 on ep 0 out, max: 8
Last buf (0 bytes left)

*** PANIC ***

ep 0 out was already available

```

lurch, 2021-04-26:
Duplicate of https://github.com/raspberrypi/tinyusb/issues/6 ?