raspberrypi/pico-sdk #363: getchar() hangs in RELEASE compiling mod, not in DEBUG compiling mode

martinkooij, 2021-04-26:
In my repo https://github.com/martinkooij/pi-pico-getchar there is a short bit of code showing a program that hangs when entering characters via **getchar().** 

The use of this call causes the pico to hang at random, but mostly before having entered 10 characters. This happens when compiled in  RELEASE mode. 
 - When using the -DCMAKE_BUILD_TYPE=Debug flag in CMake it does work flawlessly. 
 - When explicitly using a tight loop with getchar_timeout_us(0) it also works without problem.

lurch, 2021-04-26:
Looking at [getchar](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio/stdio.c#L258) I see that it uses `at_the_end_of_time`, and I remember there being some problem with that a while ago... https://github.com/raspberrypi/pico-sdk/pull/145 or https://github.com/raspberrypi/pico-sdk/issues/255

Is your pico-sdk up to date?

martinkooij, 2021-04-26:
Ahhh. Alas, I cloned a fresh pico-sdk this weekend (24 April 2021).

kilograham, 2021-04-26:
could not repro; pls attach a UF2 (and the .DIS file)

martinkooij, 2021-04-26:
[zipper.zip](https://github.com/raspberrypi/pico-sdk/files/6380236/zipper.zip)
Didn't reproduce either anymore. Don't know why. But was able to reproduce surefire with a somewhat more complicated example. Attached that one, cmakefile, .cpp files and the resulting DIS and UF2 at my computer. Fails at first try here with getchar(). Works with mygetchar().. Hope that helps, thanks for looking into it.

kilograham, 2021-04-27:
> Didn't reproduce either anymore. Don't know why. But was able to reproduce surefire with a somewhat more complicated example.
Yeah, one part of the problem was that the linker was getting to pick between two different implementations of `getchar` arbitrarily, however there was an actual bug beyond that.

I assume you are using gcc 10 as I couldn't repro it on anything earlier.

martinkooij, 2021-04-27:
Yes, gcc 10.2 here.

kilograham, 2021-05-04:
merged

PR #364 (2021-05-04): Fix possible IRQ worker starvation in release gcc 10 builds in stdio_usb
Fixes #363 

Also adds missing linker wrap argument for `getchar` which was making the build non deterministic (either our or _newlib_ 's `getchar` would be used)
closed 2021-05-04 by kilograham