raspberrypi/pico-sdk #368: clock_configure() doesn't remember the real frequency

Reneg973, 2021-05-01:
just the input frequency is stored in `configured_freq`.
Therefore please change to:
```
    // Store the configured frequency
    configured_freq[clk_index] = (uint32_t)(((uint64_t) src_freq << 8) / div);
```

lurch, 2021-05-01:
Your message is lacking context, so I assume you're talking about [this](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_clocks/clocks.c#L42) function?

Is the concern here the "real frequency" differing from the "requested frequency" because of the 24.8 fractional divider?
Rather than change the line where the `configured_freq` is updated, perhaps it makes sense to add:
```
// Calculate the actual frequency that will be used
freq = (uint32_t) (((uint64_t) src_freq << 8) / div);
```
immediately after https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_clocks/clocks.c#L51 ? :man_shrugging:

Reneg973, 2021-05-01:
Of course of course, I thought when I write "clock_configure()" as title and there is just one function with this name in the complete SDK, the context is clear? :)
Yes fixing it this way would make it more readable. The result will be the same, namely that clock_get_hz() returns a the frequency which is set

commit 074633e7 (2021-05-24): store actual clock frequency in clock_configure (fixes #368)
commit 3f420054 (2021-05-27): store actual clock frequency in clock_configure (fixes #368)
commit 5afa3636 (2021-06-02): Small API additions and minor fixes (#406)

* Add missing DREQ_s

* store actual clock frequency in clock_configure (fixes #368)

* use dma DREQ values defined in dreqs/dma.h

* Fix hw_is_claimed, and add xxx_is_claimed APIs

* Add some PIO irq helper methods

* Add DMA channel IRQ status getter and clear methods

* Implement the correct PIO IRQ status/clear methods (good to have methods here as the h/w interrupt registers are super confusing)

* fix pico_multicore dependencies

* add missing wrapper func __aeabi_f2d

* Further DMA/PIO IRQ API cleanup (and review fixes)

* add PICO_INT64_OPS_I [... cut]
closed 2021-06-03 by kilograham