raspberrypi/pico-sdk #369: RTC initialization failure

Reneg973, 2021-05-01:
When running the hello_rtc example, I first wondered that there was some strange output at the end of the datetime string.
Then after reset of Pico I saw that the first value which is printed has the wrong value.

I assume this is because RTC had no time to load the given values. I don't know whether there is a bit to wait on.
For me it was enough to call `busy_wait_us(50);` after `rtc_set_datetime()`. 
On the other hand the use case to read the rtc short after it was reinitialized is very low. 
Do you think this should be fixed? If not in SDK, than at least in the example code

lurch, 2021-05-01:
> I assume this is because RTC had no time to load the given values.

The RP2040 datasheet mentions "All RTC register reads and writes are done from the processor clock domain clk_sys. All data are synchronised back and forth between the domains. Writing to the RTC will take 2 clk_rtc clock periods to arrive, additional to the clk_sys domain. This should be taken into account especially when the reference is slow (e.g. 1 Hz).", so maybe that's not being done for the `hello_rtc` example? :shrug: 
Feel free to submit a PR to `pico-examples` if you manage to get it working properly :slightly_smiling_face:

Reneg973, 2021-05-01:
ok PR submitted, closed

closed 2021-05-01 by Reneg973