raspberrypi/pico-sdk #387: fails to generate project(name C)

Wirtos, 2021-05-07:
In default examples, project has no explicitly used languages, which implies C and CXX LANGUAGES, if I set my project to be C only (which is correct), cmake fails to configure it. I think toolchain's internals should be separated from the projects like VCPKG does. 

Error log:
```
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CXX_LINK_EXECUTABLE

CMake Generate step failed.  Build files cannot be regenerated correctly.

Problems were encountered while collecting compiler information:
	cc1plus: fatal error: /home/wirtos/GithubProjects/pico-sdk/src/common/pico_stdlib/include: No such file or directory
	cc1plus: fatal error: /home/wirtos/GithubProjects/pico-sdk/src/rp2_common/hardware_gpio/include: No such file or directory
	cc1plus: fatal error: /home/wirtos/GithubProjects/pico-sdk/src/common/pico_base/include: No such file or directory
	cc1plus: fatal error: /mnt/c/GithubProjects/picolrn/cmake-build-debug-wsl-gcc/generated/pico_base: No such file or directory
	cc1plus: fatal error: /home/wirtos/GithubProjects/pico-sdk/src/boards/include: No such file or directory
 ...
```

lurch, 2021-05-07:
Which part of your project was being built when you saw those errors? Note that building the toolchain also builds [elf2uf2](https://github.com/raspberrypi/pico-sdk/tree/master/tools/elf2uf2) (which converts the `.elf` file that GCC generates into a `.uf2` file for copying onto your Pico) which is a native (it runs on your computer rather than on your Pico) program written in C++.

Wirtos, 2021-05-07:
> Which part of your project was being built when you saw those errors? Note that building the toolchain also builds [elf2uf2](https://github.com/raspberrypi/pico-sdk/tree/master/tools/elf2uf2) (which [...]

It breaks before the build process during generation as mentioned in the report. Does this regardlessly of me having add_extra_outputs. The toolchain config should be responsible for providing C++ compiler to build elf2uf2. Also when looking at the elf2uf2 code it looks like it mustn't be necessarily C++, but I'm not sure that elf2uf2 is the root of the problem.

kilograham, 2021-05-08:
Yeah; fixing this is a bit nasty, because the SDK itself is a CXX project, and reports as such vi ENABLED_LANGUAGES property in cmake. Your project fails, because at the toplevel scope CXX is not a valid language. I had a little play and see that if the SDK doesn't include new_delete.cpp then you are fine, but as I say we can't simply check ENABLED_LANGUAGES as that does contain CXX at this point.

I don't feel strongly enough to expend much effort on it right now, so feel free to post a PR... I guess we could add a CMake build flag at least to say you want nothing to do with C++ assuming it is an actual requirement

See #339 for C99

Wirtos, 2021-05-09:
I guess I'll try to toy with the toolchain later and maybe port tools like elf2uf2 to C then

kilograham, 2021-05-09:
elf2uf2 is _not_ the problem; that is a separate build and works fine

Wirtos, 2021-05-09:
So it's solely the toolchain that breaks because of undefined languages? It seems like it's quite trivial to use gcc-arm in place of CMAKE_TOOLCHAIN_FILE, but I'm not sure on how to link the libraries in this case. iirc vcpkg adds directories to CMAKE_MODULE_PATH in order to be able to use find_package or find_path. Can I do the same separately for pico toolchain?

Bolayniuss, 2021-07-08:
I manage to solve this by using a variable `PICO_SDK_PURE_C` to remove `new_delete.cpp` from the source.

In `src/rp2_common/pico_standard_link/CMakeLists.txt` replace 
``` C
target_sources(pico_standard_link INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}/crt0.S
    ${CMAKE_CURRENT_LIST_DIR}/new_delete.cpp
    ${CMAKE_CURRENT_LIST_DIR}/binary_info.c
)
```
by
``` C
set(pico_standard_link_SOURCES ${CMAKE_CURRENT_LIST_DIR}/crt0.S ${CMAKE_CURRENT_LIST_DIR}/binary_info.c)
if(NOT DEFINED PICO_SDK_PURE_C)
    message("C++ enabled, add ${CMAKE_CURRENT_LIST_DIR}/new_delete.cpp")
    set(pico_standard_link_SOURCES ${pico_standard_link_SOURCES} ${CMAKE_CURRENT_LIST_DIR}/new_delete.cpp)
endif()
```
And then, in your `CMakeList.txt` define add `set(PICO_SDK_PURE_C 1)` before `include(pico_sdk_import.cmake)`

JannickBremm, 2021-08-04:
> I manage to solve this by using a variable `PICO_SDK_PURE_C` to remove `new_delete.cpp` from the source.
>
> In `src/rp2_common/pico_standard_link/CMakeLists.txt` replace
>
> ```c [...]

Worked for me! Thank's a lot @Bolayniuss