raspberrypi/pico-sdk #396: Support building precompiled libraries for users of other toolchains

clausecker, 2021-05-11:
The build system should support building and installing the `elf2uf2` and `pioasm` utilities, documentation as well as pre-compiled library files for users of toolchains other than CMake.  Right now, it is kinda hard to even coax the build system into producing these binaries separately.  The goal here is to support making binary packages of the SDK for easy installation through the distribution's package manager as well as making it easier to build RPi Pico projects without having to fight with CMake.

clausecker, 2021-05-11:
> Who's goal is this? Is this easier than checking out a single github project, or having the SDK as a submodule

My goal.  I do not want to rebuild the entire standard library for every project or even use CMake for my own projects.  I am not a fan of CMake; this is not a matter of refusing to learn it, but rather about things like integration into existing projects that already use a different build system or have special requirements.  It is also about giving the user the choice to select a build system instead of forcing CMake onto everybody.

> You would still have to build them once; separate installation packages across all supported platforms is still a lot of extra work.

That is correct.   But once built and installed, the tools and libraries can be used with custom build systems, giving the programmer more flexibility.  I also kinda don't understand the point of wasting tons of space rebuilding the library and tools for each and every project.  Seems kinda wasteful.

I do not ask the project to provide binary packages.  Merely to make it so that generating binary packages is possible, e.g. by adding targets to build all utilities and to precompile the standard library for a given board (let the distribution worry about building the library for each board separately).  A distribution can then pick up the toolkit and make a package.

clausecker, 2021-05-11:
Thank you for the answer.

> On elf2uf2 etc, these can be built separately from within their own build directory

Yes; though slightly inconvenient, this is workaround.  Have these built by default when building the SDK on top level (as opposed to, as a dependency) would already be a nice improvement.  Also, an `install` target to install at least these tools and the documentation could be helpful.  The project already has logic to check for the tools in the user's `PATH` it seems and they don't depend on the project configuration at all, so compiling them once ahead of time seems useful enough.

> You certainly aren't forced to use CMake, but it is the only supported way.

If it's the only supported way it seems a lot like I am pretty much forced to use CMake.  I don't really get why the traditional static library approach doesn't work for you; perhaps there are some shenanigans with generated or conditional code that I haven't found yet (didn't dug into the `src` directory too deeply).  And I don't get how recompiling each module for every project is supposed to be an advantage, but I understand if you don't want to argue this point.

Arch “packaged” the SDK by just shipping the source.  But even that seems dubious as they don't initialise any submodules of `tinyusb` and it is unclear which are needed for which configurations.  Shipping all of them would be too much wasted space, clocking in at almost 400 MB (there is stuff like a full rtos tree in there).

The pico-playground link with the static library you posted does seem like it is useful.  I'll go ahead and give it a try.

> Note, running the tool above will show just how many things you weren't having to worry about in terms of command line defines etc.

I do see you use a lot of `-Wl,--wrap=...` options; it honestly doesn't seem to be too bad.

As for some background: we've been porting mecrisp-stellaris, a Forth system, to the Pico board.  The build process is a bit special with a custom linker script, custom boot loader, custom runtime and ABI, and an intermediate step where the image is loaded into an emulator to compile additional definitions in.  Right now, we use a conversion script from Microsoft's UF2 tools and have copied all magic constants for the peripherals in by hand.  It might be useful to make use of some parts of the SDK (such as pin definitions) to allow for reuse of the port to other boards, reduce the number of magic constants as well as to embed metadata into the image.  But integrating CMake into the build process of a project that already builds images for dozens of other boards without requiring such dependencies seems to be a lot of work and added complexity for not really much of a gain.

lurch, 2021-05-11:
> And I don't get how recompiling each module for every project is supposed to be an advantage, but I understand if you don't want to argue this point.

Just for an example, have a look at how many compile-time settings https://github.com/raspberrypi/pico-project-generator allows you to set. A lot of the configuration steps (?) that you might have to do with a statically-built library, are instead done at compile time leading to more optimised executables.

Each time you run cmake it doesn't rebuild the entire standard library, it only compiles the parts of the code actually used by your project. It's all very clever :wink: (Magic @kilograham stuff that I don't fully understand myself yet!)

> The build process is a bit special with a custom linker script, custom boot loader, custom runtime and ABI, and an intermediate step where the image is loaded into an emulator to compile additional definitions in.

LOL, that sounds a bit more complicated than just following the instructions in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf and running cmake? :wink: 

> If it's the only supported way it seems a lot like I am pretty much forced to use CMake.

I think Graham might have been implying that whilst CMake is the only build system that we'll be supporting in the SDK, the entire SDK sourcecode is BSD-licensed, so you're free to wrap your own build-system around the SDK's source-files?

This issue reminds me a lot of #17 

>  It might be useful to make use of some parts of the SDK (such as pin definitions) to allow for reuse of the port to other boards

See https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards ?

kilograham, 2021-05-11:
> But I'm going to duck out of this conversation now because there's not really anything useful I can add.

yes... not really very helpful

clausecker, 2021-05-12:
> This would have been good context up front :-)
>
> Yes, having the tools build as part of the build, rather than installation, was a) good for bootstrapping, but more importantly b) a lot simpler when you need to support Linux, Mac, and Windows (both with/without cygwin/wsl/wsl2) - always worth keeping that in the back of your mind when considering all these things.
> 
> Having an install option totally makes sense, and please open a separate issue.

If you like, I can make this issue about the install option.  The Forth system was more of an example for how not every project can realistically be structured around the Pico SDK's demands on a build system.

> I am out of the loop of anyone packaging SDK. tinyusb has a horrible swamp of submodules (fortunately none are needed for pico_sdk), however they are removing all of them I think in a subsequent release.

Ah, that sounds like good news at least.

> The complicating factor with RP2040 is that you must prepend a "2nd stage" flash onto the binary, since the RP2040 has no knowledge of the type of flash in use - this boot stage 2 is dependent on the type of flash used (though you can use a slow generic one). pico_standard_link CMake library does this for you,

Correct.  For Mecrisp Stellaris, we actually wrote a custom boot loader for this purpose.

> If you are doing custom linking stuff anyway, then yes you need to adapt our linker script. Again yes, the link scripts are not well documented, but such things are way off the happy path - that is on the list though (along with templateization)

Sounds good!

> Anyways, for your use case what does the SDK get you possibly? Most likely of use:

Specifically, I wanted to get macros for register definitions and the included binaries.  Especially `pioasm` might be useful for some projects.

> elf2uf2 is not really required; you can take the .BIN and pass it thru Microsoft's python tool

Yes, that's what we do right now.

> cleverness was not my word, nor the goal. We do not test on FreeBSD, if you open an issue with the error message that may help identify the cause.

Yes, @lurch said that, not you.  I'll see if I can open an issue about that.

aallan, 2021-05-12:
> > We do not test on FreeBSD, if you open an issue with the error message that may help identify the cause.
> 
> I'll see if I can open an issue about that.

I've got a vague recollection that there is an open (closed) issue already, but for the life of me I can't find it. Either here, or on the raspberrypi/pico-feedback repo.

lurch, 2021-05-12:
> I wanted to get macros for register definitions

The register definitions are all nicely self-contained in https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2040

> > re: FreeBSD
> I've got a vague recollection that there is an open (closed) issue already, but for the life of me I can't find it.

I can't find it either: https://github.com/search?o=desc&q=org%3Araspberrypi+freebsd&s=updated&type=Issues
Maybe it was something that only got mentioned during the closed beta-testing? Or maybe it was mentioned on the forums but never got as far as a GitHub issue? :shrug:

aallan, 2021-05-12:
> Maybe it was something that only got mentioned during the closed beta-testing? Or maybe it was mentioned on the forums but never got as far as a GitHub issue? 

Ah ha! It was a [forum post](https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=311022).

arg08, 2021-05-15:
> > (btw, I just noticed the SDK doesn't even build on FreeBSD because it tries to pull in system C++ headers that get very confused by the embedded environment and complain loudly; so much for “very clever”).
> 
> _cleverness_ was not my word, nor the goal. We do not test on FreeBSD, if you open an issue with the error message that may help identify the cause.

I'm pretty sure that this is a defect in the FreeBSD port/package for arm-none-eabi-gcc rather than anything to do with the Pico SDK or Cmake.  Fortunately it's easily worked around by using gcc-arm-embedded instead.

I got as far as finding where the build was breaking and running by hand the command that the build system was doing at that point:

```
% arm-none-eabi-g++ -c /home/arg/work/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp
In file included from /usr/include/c++/v1/cstdlib:84,
                 from /home/arg/work/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:10:
/usr/include/c++/v1/__config:1132:6: error: #error "No thread API"
 #    error "No thread API"

```
Where line 10 is just #include <cstdlib>

So the problem is that the g++ cross-compiler is incorrectly picking up host include files.  Pico-sdk doesn't seem to be doing anything clever or complex here.

clausecker, 2021-05-15:
@arg08 Thanks, I'll go ahead and file a bug report with the port.

lurch, 2023-02-12:
Does #1221 partially address this?

peterharperuk, 2024-05-08:
Removed unhelpful comment.

will-v-pi, 2024-08-09:
This should be fixed with SDK 2.0.0 - the elf2uf2 functionality has been moved into [picotool](https://github.com/raspberrypi/picotool), and pioasm now installs in a way that builds using the SDK will be able to locate and use it. This installs it both to the PATH, and in a way that CMake `find_package` can locate it, so you can use either of those methods to integrate it into your own build system.

To install pioasm, you can run
```
cd tools/pioasm

mkdir build
cd build
cmake ..
make
sudo make install
```

closed 2024-08-09 by will-v-pi