raspberrypi/pico-sdk #403: flash_range_erase() granularity

dshadoff, 2021-05-12:
The SDK documentation webpages (v1.1.12) describe flash_range_erase() as follows:

void flash_range_erase ( uint32_t  flash_offs, size_t  count )
flash_offs - Offset into flash, in bytes, to start the erase. Must be aligned to a 4096-byte flash sector.
count - Number of bytes to be erased. Must be a multiple of 4096 bytes (one sector).

However, the implementation at pico-sdk/src/rp2_common/hardware_flash/flash.c implements the SPI command:
#define FLASH_BLOCK_ERASE_CMD 0xd8
...which erases by block (65536 bytes).  (The sector erase command would be 0x20, or also 0xd7 on the PICO-specific chip)

It's not clear how widely the BLOCK or SECTOR-based SPI erase commands would be implemented across other vendors' SPI Flash devices, but I believe that SECTOR is likely the more commonly-implemented command.

If this is deemed to be a SDK documentation issue, the corresponding pico-examples code should be amended to agree.

lurch, 2021-05-12:
Flash-erasing is outside my area of knowledge, but you may want to also look at https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/program_flash_generic.c#L308

dshadoff, 2021-05-12:
Thank you, this is very helpful !

In truth, this issue wasn't causing me problems, as I was able to adapt my code; I just wanted to ensure that the discrepancy between documentation and implementation was known, and would be addressed at some point.

kilograham, 2021-05-12:
yeah it seems like we should be using the `#define FLASHCMD_SECTOR_ERASE     0x20` command used in the bootrom. This is likely a @Wren6991 question, but maybe just that code originated somewhere that liked to erase quickly!

lurch, 2021-05-12:
> as I was able to adapt my code

What did you need to "adapt"? AFAICT from a cursory glance at the code, it uses `FLASH_BLOCK_ERASE_CMD` for `FLASH_BLOCK_SIZE` sized (and aligned) chunks, and `FLASHCMD_SECTOR_ERASE` for anything else? Presumably because erasing a block (if possible to do so) is quicker than erasing many separate sectors?
But it _looks_ like this is just a bootrom implementation detail, and the API presented by the SDK remains the same?
Also, see section 2.8.2.1.3. of the RP2040 datasheet.

@kilograham Does the SDK maybe need to provide PICO_CONFIG defines for `FLASH_BLOCK_SIZE` and `FLASH_BLOCK_ERASE_CMD` so that the https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/flash.c#L63 function works for "other" flash chips? :man_shrugging:

kilograham, 2021-05-12:
> @kilograham Does the SDK maybe need to provide PICO_CONFIG defines for `FLASH_BLOCK_SIZE` and `FLASH_BLOCK_ERASE_CMD` so that the https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/flash.c#L63 function works for "other" flash chips? man_shrugging

Not sure here, though potentially a fast erase cmd/size might be helpful for users who want to erase as fast as possible

dshadoff, 2021-05-12:
> > as I was able to adapt my code
> 
> What did you need to "adapt"?

I am working with a 128KB block of memory, and I decided not to break it into 4KB sectors...

> AFAICT from a cursory glance at the code, it uses `FLASH_BLOCK_ERASE_CMD` for `FLASH_BLOCK_SIZE` sized (and aligned) chunks, and `FLASHCMD_SECTOR_ERASE` for anything else? Presumably because erasing [...]

Mmm... maybe you can see more than I can, but I don't understand the mechanism of how the SDK would reference such bootrom functionality (and I did not dive into the bootrom code to examine it).  I saw only the literal BLOCK level command when searching the pico-sdk repo.

And yes, erasing a block is considerably faster than erasing a series of sectors.

lurch, 2021-05-13:
> Mmm... maybe you can see more than I can, but I don't understand the mechanism of how the SDK would reference such bootrom functionality

So...
* The SDK-level function is [`flash_range_erase(uint32_t flash_offs, size_t count)`](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/flash.c#L63) (which matches up with the [SDK documentation](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__flash.html#ga5b88461b126b1c226121146c87d5abe2) )
* This then uses `rom_func_lookup` to create a (private) function-pointer [`flash_range_erase(uint32_t, size_t, uint32_t, uint8_t)`](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/flash.c#L71) (yeah, it's a little bit confusing that the SDK-function and the function-pointer are both named `flash_range_erase` :wink: )
* This function-pointer is the `'R','E'` bootrom-function described in section 2.8.2.1.3. of the [RP2040 datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf).
* And this bootrom-function is itself implemented [here](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/program_flash_generic.c#L333) (also called `flash_range_erase`!). And in there you can see the logic where it drops through to calling [`flash_sector_erase`](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/program_flash_generic.c#L340) if the chunk being erased isn't block-aligned or is smaller than a block.

Of course this is only how I understand it (I didn't write any of this), and hopefully @kilograham will point out any errors I've made in my description!

kilograham, 2021-05-13:
look; as I said, @Wren6991 is the one who would know the reason for this if any, but the point remains the same as pointed out by @dshadoff; the documentation for the function does **not** match the implementation. And as he says the implementation using 4K probably makes more sense as it is more likely supported.

kilograham, 2021-05-13:
I removed my previous comment as this is actually confusing; and I think all the right things have been said...

1) The documentation _is_ correct... the data must be 4K aligned, and a multiple of 4K in size.
2) The implementation of the SDK function `flash_range_erase` uses `FLASH_BLOCK_ERASE_CMD (0xd8)` - whether this is safe across most flashes is a question for @Wren6991 and we could opt to add a board config value to specify

It is fair to say that the _bootrom_ function called _flash_range_erase is a confusing beast... the arguments are a bit overloaded in a weird way to save code space.

dshadoff, 2021-05-13:
To be clear, I haven't demonstrated that it isn't working as documented; while solving my problem with getting a program to work properly, I was looking into this function and saw an apparent mismatch.  If the bootrom code is indeed referenced as @lurch explains (and I couldn't see), then I am wrong and this is a red herring.

dshadoff, 2021-05-14:
I have run tests and confirmed that it does function as per documentation and that this was my misunderstanding.
I am fine with having this issue closed, but will leave it to @kilograham / @Wren6991 due to the question posed by kilograham still being open.

Wren6991, 2021-05-17:
I think everything has been said, but trying to summarise:

- The underlying ROM function `flash_range_erase()` requires its target region to be 4k-aligned and sized to a multiple of 4k
- The ROM function requires the flash to implement the `20h` 4k-erase command
- The ROM function accepts as parameters an additional command that your flash supports (this could be anything up to and including a chip erase command), and the size of the naturally-aligned block erased by this command
- The ROM function uses the additional command is where possible (assumed to be larger and faster), and falls back on the `20h` 4k erase for smaller or unaligned parts of the erase region
- The *SDK* passes in the `d8h` 64k-erase command when calling the ROM function, for better performance when erasing large regions
- When called by the SDK, the ROM function uses a mixture of `20h` and `d8h` commands to erase the target region, depending on size and alignment

`d8h` 64k erase is very widely supported, but we would be happy to add some defines if it turns out to be an issue.

closed 2021-05-17 by Wren6991