raspberrypi/pico-sdk #409: error in flash size for Adafruit Feather rp2040 and ItsyBitsy rp2040?

jgriesfeller, 2021-05-14:
Hi,
 
while  trying to add the Adafruit Feather rp2040 and ItsyBitsy rp2040 boards to Micropython I noticed a difference in flash size in the board definitions at `pico-sdk/src/boards/include/boards/adafruit_itsybitsy_rp2040.h` and `pico-sdk/src/boards/include/boards/adafruit_feather_rp2040.h`.
According to the web site [feather rp2040](https://www.adafruit.com/feather2040) and [ItsyBitsy rp2040](https://www.adafruit.com/itsybitsy2040) they have 8MB flash, but the board definitions (in the master branch at least) have just 4MB defined :
`#define PICO_FLASH_SIZE_BYTES (4 * 1024 * 1024)`

Is this a leftover from pre-production samples that did not have the 8MB flash?

I do have the boards but did not try to write more than 4MB to the flash yet, but Circuitpython shows the additional 4MB as present.

lurch, 2021-05-14:
This was fixed in #348

fivdi, 2021-05-14:
For the Adafruit Feather RP2040 `PICO_FLASH_SIZE_BYTES` is defined as 8 MB in [master](https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/adafruit_feather_rp2040.h#L79-L81) and in [pico-sdk v1.2.0](https://github.com/raspberrypi/pico-sdk/blob/1.1.2/src/boards/include/boards/adafruit_feather_rp2040.h#L79-L81).

For the Adafruit ItsyBitsy RP2040 `PICO_FLASH_SIZE_BYTES` is defined as 4 MB in [master](https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/adafruit_itsybitsy_rp2040.h#L82-L84) and in [pico-sdk v1.2.0](https://github.com/raspberrypi/pico-sdk/blob/1.1.2/src/boards/include/boards/adafruit_itsybitsy_rp2040.h#L82-L84). This was fixed in the [develop branch](https://github.com/raspberrypi/pico-sdk/blob/develop/src/boards/include/boards/adafruit_itsybitsy_rp2040.h#L82-L84) with PR #293.

fivdi, 2021-05-14:
What MicroPython sees will depend of the version of pico_sdk it uses.

jgriesfeller, 2021-05-14:
I admittedly did not search through the dev branch. Sorry for the clutter.

closed 2021-05-14 by jgriesfeller