raspberrypi/pico-sdk #412: cannot access XIP from sram build

nonchip, 2021-05-16:
while `hardware_flash` provides functions to write/erase the flash when running from sram (type `no_flash`), i'd need to both read from and execute into XIP, but sram builds don't (seem to) provide a way of enabling `boot_stage2` and the sdk doesn't (seem to) provide any other function to initialize the flash for XIP.

manually adding `boot_stage2` as a `target_link_library` fails too because then the linker complains about there not being a `libboot_stage2.so`

could you please provide a "load into and run from sram" mode that doesn't disable flash access (= includes `boot_stage2`)?

lurch, 2021-05-16:
Is the `_flash_enter_cmd_xip` bootrom function (detailed in section 2.8.2.1.3. of the [RP2040 datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) what you're looking for? :man_shrugging:

kilograham, 2021-05-16:
there is `copy_to_ram` as well as `no_flash`

nonchip, 2021-05-16:
@lurch i thought so, and i tried it like this:
```C
int main() {
  stdio_init_all();

  void (*flash_enter_cmd_xip)(void) = (void(*)(void))rom_func_lookup(rom_table_code('C', 'X'));
  assert(flash_enter_cmd_xip);
  flash_enter_cmd_xip();
  // rom_funcs below is *not* the bootrom stuff but a const struct defined in the XIP .rodata:
  while (true) {
    printf("rom_funcs* %x\n",&rom_funcs);
    printf("rom_funcs: %x\n",rom_funcs);
    printf("     init: %x\n",rom_funcs.init);
    printf("       on: %x\n",rom_funcs.on);
    printf("      off: %x\n",rom_funcs.off);
    printf("\n");
    sleep_ms(1000);
  }
}

```

but the assertion fails.

@kilograham i know and that would be the exact opposite of what i need, given i *don't* want to overwrite the flash before accessing it.

lurch, 2021-05-16:
> (i don't have a SWD probe that can debug sadly)

You're _probably_ already aware of this, but just in case you're not: You can use either a Raspberry Pi or a second Pico for this. Instructions in the [Getting Started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf).

nonchip, 2021-05-16:
@lurch yeah i am but the only pi i have currently is hardwired into my 3d printer and i don't have a 2nd pico yet (currently stuck in shipping), so i was hoping anyone could tell me if i'm just calling that function wrong until it arrives

nonchip, 2021-05-16:
in case anyone wants to see the full code: https://gitlab.com/nonchip/pi-pico-xipram/-/blob/master/ram/ram.c is where i do the xip access, `../xip/xip.c` is where things are declared, everything is supposed to be built using the top level cmakelists (kinda like the pico-examples), and if it worked it should be doing the same as the blink after loading both `uf2`s (the xip one resets into bootloader, so both after flashing it and after powering up it's ready for the ram one)

kilograham, 2021-05-16:
It seems like it should work, so I tried your code, and it works fine for me. I'm not sure what you're trying to do; the code in the bootrom will get you generic (slower speed) access to XIP.

The simplest thing to do might be to have a valid program in flash which just calls into RAM, then you can run boot stage2 by doing a watchdog reboot - note also it is possible to call the second stage boot as a function (you have to copy it into RAM at the right place first); i could swear there was an example that does this, but I cannot find it atm.

kilograham, 2021-05-16:
> what actually happens is as soon as i load the ram uf2 it initializes cdc and hangs.

oh, i didn't read this - i have UART hooked up, and saw the right stuff there. might be worth trying it without a USB terminal hooked up to see if it makes a difference.

nonchip, 2021-05-16:
> It seems like it should work, so I tried your code, and it works fine for me.

i just cleared my whole `build` folder and built it again just in case there was some old object not being built there, `dmesg` says:
```
[290103.139857] usb 1-12: device descriptor read/64, error -110   // first reboot after XIP flash
[290119.012858] usb 1-12: device descriptor read/64, error -110   // 2nd reboot after RAM loaded
[290119.236802] usb 1-12: new full-speed USB device number 64 using xhci_hcd // CDC initialized
[290134.883853] usb 1-12: device descriptor read/64, error -110  // immediately disconnects and stays dead
```

> I'm not sure what you're trying to do; the code in the bootrom will get you generic (slower speed) access to XIP.

essentially what i was trying to do is have a bunch of low level common functionality i'll need for a "modular" kinda project in XIP, then load application specific code into RAM which accesses that to save on code duplication and flash destruction.

> The simplest thing to do might be to have a valid program in flash which just calls into RAM

yeah that's what i thought first but then i'd have to implement the bootloader for the RAM code loading myself so i figured i'd rather have the program in flash reboot into the bootloader :/

nonchip, 2021-05-16:
> might be worth trying it without a USB terminal hooked up to see if it makes a difference.

inspired by this i tried putting the stdio init *below* the XIP init stuff, now i get this:
```
[290505.251783] usb 1-12: new full-speed USB device number 69 using xhci_hcd
[290517.378796] usb 6-2: Device not responding to setup address.
[290517.585826] usb 6-2: Device not responding to setup address.
[290517.793791] usb 6-2: device not accepting address 91, error -71
[290517.970862] usb 6-2: Device not responding to setup address.
[290518.178800] usb 6-2: Device not responding to setup address.
[290518.385777] usb 6-2: device not accepting address 92, error -71
[290520.931852] usb 1-12: device descriptor read/64, error -110
[290536.805835] usb 1-12: device descriptor read/64, error -110
[290537.027786] usb 1-12: new full-speed USB device number 70 using xhci_hcd
[290552.675829] usb 1-12: device descriptor read/64, error -110
[290568.547832] usb 1-12: device descriptor read/64, error -110
[290568.649835] usb usb1-port12: attempt power cycle
[290569.277786] usb 1-12: new full-speed USB device number 71 using xhci_hcd
[290574.103859] usb 1-12: Device not responding to setup address.
[290579.131986] usb 1-12: Device not responding to setup address.
[290579.337801] usb 1-12: device not accepting address 71, error -71
[290579.451800] usb 1-12: new full-speed USB device number 72 using xhci_hcd
[290584.278086] usb 1-12: Device not responding to setup address.
[290589.308189] usb 1-12: Device not responding to setup address.
[290589.513778] usb 1-12: device not accepting address 72, error -71
[290589.513832] usb usb1-port12: unable to enumerate USB device
```

kilograham, 2021-05-16:
yeah using USB as a starting point is likely making life hard.... note for a start you are likely to be shooting yourself in the foot if your flash based "program" uses any static mutable data, as without extra work it is going to stomp on the data of the ram based program.

> yeah that's what i thought first but then i'd have to implement the bootloader for the RAM code loading myself so i figured i'd rather have the program in flash reboot into the bootloader :/

Not sure what RAM bootloader you are talking about?

nonchip, 2021-05-16:
> as without extra work it is going to stomp on the data of the ram based program

oh yeah that's something to keep in mind, i'd have to allocate any buffers in the ram program i guess

> Not sure what RAM bootloader you are talking about?

well if i have code sitting in XIP which you suggested should jump to ram instead of just resetting to the bootloader like i'm currently doing, then i'd have to make that XIP code actually load the code to ram, because the builtin bootloader isn't gonna do that for me then.

nonchip, 2021-05-16:
to elaborate on what i'm trying to do: build a "smart" adapter for a variety of different serial interfaces (Gameboy/GBA Link to be precise), where i'll put the low level "phy" layer stuff that never changes (and common features like the GBA wireless adapter emulation, that don't depend on the game being linked up to) in XIP, and then have an application on the host (the USB host, aka PC, not the gameboy) that can load game-specific logic onto RAM which then interfaces that to the application over USB, because there's various games that require handshakes / ping timeouts (all implemented differently without any standard) faster than even USB can guarantee.

nonchip, 2021-05-16:
so i'll need to:
- either boot to XIP and then somehow provide a way to load code onto RAM which gets called from the XIP code (which i'd kinda prefer but don't know how to do without having to reinvent the wheel that is bootloading and linking/crt)
- or boot to RAM (which i'm currently doing, note my XIP code literally just resets into the bootloader) and call into functions in XIP; but which seems to fail somewhere along "using USB and XIP at the same time" (which makes no sense to me)

kilograham, 2021-05-16:
> but that sounds very much like a load of pain and boils down to "please gimme a target mode that runs in ram that isn't no_flash" again :P

well someone's pain, and given how much other stuff we have going on, probably gonna be yours :-)

> well if i have code sitting in XIP which you suggested should jump to ram instead of just resetting to the bootloader like i'm currently doing, then i'd have to make that XIP code actually load the code to ram, because the builtin bootloader isn't gonna do that for me then.

I was suggesting you make a flash binary, which when run just jumps into a known address in the RAM binary (in fact just the beginning of binary is fine with a RAM binary). Your RAM binary when run checks one of the watchdog scratch registers to see what to do; (either do a watchdog reboot (into flash) having set a value in the scratch register for the next re-entry into the RAM binary), or continue executing and clear the special value.

nonchip, 2021-05-16:
> I was suggesting you make a flash binary, which when run just jumps into a known address in the RAM binary

ooh that might actually work, especially given i can "flash" the ram binary using `picotool` and the magic baudrate without having to ever actually reset into the bootloader for the usb drive emulation.

good idea and sounds rather easy to achieve, will try :)

kilograham, 2021-05-16:
that works too; though i was just suggesting you run the RAM binary, and it reboots into flash, which then calls back into RAM

nonchip, 2021-05-16:
@kilograham ok that worked, ...somewhat.

```C
// XIP:
int main() {
  if(watchdog_caused_reboot()){
    int (*ram_entry)() = (int(*)())0x20000000;
    return ram_entry(); // jump to ram
  }
  reset_usb_boot(0, 0); // reboot into bootloader
}

// RAM:
int main() {
  if(!watchdog_caused_reboot()){
    int (*xip_entry)() = (int(*)())0x10000000;
    watchdog_reboot((uintptr_t)xip_entry,SRAM_END,10);
    while (true); // let the watchdog kill us
  }
  stdio_init_all();
  /// my printf loop
}
```

problem is my output is now:
```
xip_funcs* 10001c20␍␊ // correct
xip_funcs: 0␍␊ // all wrong.
     init: 0␍␊
       on: 6␍␊
      off: 0␍␊
```

kilograham, 2021-05-16:
I'm sure you can debug on your own, but I will note that the ram pointer should be `0x20000001` so your probably aren't reaching there or it'd hard fault.

Also if you want to reboot into flash, you should pass 0 for the address

nonchip, 2021-05-16:
weird that it actually called into ram then, i suspect that's the compiler saving me (thumb calls are always odd). but good point for the rebooting to 0 there, trying...

nonchip, 2021-05-17:
ok yeah you were right @kilograham, something during ram initialization killed the "are we rebooting from watchdog" detection, i never actually reached the XIP code even, so my non-thumb-call didn't matter.

fixed that now by using different scratch registers, tried with both watchdog-rebooting to `ROM_BASE` and `XIP_BASE` (and both of them with and without `+1` to force thumb mode), and with having rom call into both `SRAM_BASE` (with+without `+1` again) and the address of the ram code's `main` (again even+odd), but somehow my code never reaches back to ram. guess i might have to wait for the picoprobe :/

nonchip, 2021-05-17:
what i still don't quite get is how `rom_table_code('C', 'X')` (enable slow XIP) messes with USB, maybe i can just reset usb after that was enabled?

Wren6991, 2021-05-17:
Note that the call to `flash_exit_xip()` after `connect_internal_flash()`, which you will see in the SDK flash code, is not optional. The function `flash_init_spi()` [here](https://github.com/raspberrypi/pico-bootrom/blob/00a4a19114195e20fb817bdfbca1165e157eef37/bootrom/program_flash_generic.c#L65) is folded into that call (bit of a poor API, a victim of size hacking to fit into ROM space). If that's not called following a reset of the SSI, the SSI won't be enabled, which is going to break *something* when you start accessing the flash.

To get the flash from zero to slow-XIP access the sequence of calls is

- `connect_internal_flash()`
- `flash_exit_xip()`
- `flash_enter_cmd_xip()`

lurch, 2021-05-17:
It's an entirely separate repo, but this reminds me of https://github.com/raspberrypi/openocd/pull/29

Wren6991, 2021-05-17:
> It's an entirely separate repo, but this reminds me of [raspberrypi/openocd#29](https://github.com/raspberrypi/openocd/pull/29)

Seems unrelated, this issue doesn't involve flash erase, flash write, or openocd

nonchip, 2021-05-17:
@Wren6991 omg thank you so much, i would never have guessed i need to actively exit and flush before entering, it works perfectly now!

will clean up the code a bit and put it here for future reference :)

Wren6991, 2021-05-17:
Great! The `flash_exit_xip()` call is useful anyway, because after rebooting RP2040, the external flash device may be left in a state where it won't accept serial commands like the `03h` used by slow-XIP

nonchip, 2021-05-17:
So here we go, this is the code for enabling "fast" XIP via a boot2 *inside* XIP:

```C
#include "pico/stdlib.h"
#include "pico/bootrom.h"

void __no_inline_not_in_flash_func(enable_xip_via_boot2_in_xip)(){
  void (*connect_internal_flash)(void) = (void(*)(void))rom_func_lookup(rom_table_code('I', 'F'));
  void (*flash_exit_xip)(void) = (void(*)(void))rom_func_lookup(rom_table_code('E', 'X'));
  void (*flash_flush_cache)(void) = (void(*)(void))rom_func_lookup(rom_table_code('F', 'C'));
  void (*flash_enter_cmd_xip)(void) = (void(*)(void))rom_func_lookup(rom_table_code('C', 'X'));
  assert(connect_internal_flash && flash_enter_cmd_xip && flash_exit_xip && flash_flush_cache);

  __compiler_memory_barrier();
  connect_internal_flash();
  __compiler_memory_barrier();
  flash_exit_xip();
  __compiler_memory_barrier();
  flash_flush_cache();
  __compiler_memory_barrier();
  flash_enter_cmd_xip();
  __compiler_memory_barrier();
  static uint32_t boot2_copyout[64];
  for (int i = 0; i < 64; ++i){
    boot2_copyout[i] = ((uint32_t *)XIP_BASE)[i];
  }
  __compiler_memory_barrier();
  flash_exit_xip();
  __compiler_memory_barrier();
  ((void (*)(void))boot2_copyout+1)();
  __compiler_memory_barrier();
}
```

now i only got a small issue of my test code, while apparently being found correctly, not doing what it should:
```C
// on XIP side:
void blinkinit() {
  gpio_init(LED_PIN);
  gpio_set_dir(LED_PIN, GPIO_OUT);
}

void blinkoff() {
  gpio_put(LED_PIN, 1);
}

void blinkon() {
  gpio_put(LED_PIN, 1);
}

const struct xip_funcs_t xip_funcs = {
  &blinkinit,
  &blinkoff,
  &blinkon,
};

// on RAM side:
  xip_funcs.init();
  while (true) {
    xip_funcs.on();
    sleep_ms(250);
    xip_funcs.off();
    sleep_ms(250);
    watchdog_update();
  }
```

debug output seems fine:
```
xip_funcs* 10001e64␍␊
xip_funcs: 1000031d␍␊
     init: 1000031d␍␊
       on: 10000331␍␊
      off: 10000311␍␊
```

and the led does turn *on*, but never off again :P

Wren6991, 2021-05-17:
```
void blinkoff() {
  gpio_put(LED_PIN, 1);
}

void blinkon() {
  gpio_put(LED_PIN, 1);
}
```
Hmmm :)

nonchip, 2021-05-17:
> ```
> void blinkoff() {
>   gpio_put(LED_PIN, 1);
> }
> 
> void blinkon() {
>   gpio_put(LED_PIN, 1);
> }
> ```
> 
> Hmmm :)

yeeaaaah i feel dumd, did a copypaste derp, noticed when i saw the watchdog never killing me :'D
guess this is all fixed now then :P

closed 2021-05-17 by nonchip