raspberrypi/pico-sdk #413: It is not obvious what unsetting and setting CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS in clock_configure() achieves

kahara, 2021-05-16:
(Likely reason being that I don't understand what's going on, but would still appreciate if someone who is knowledgeable enough, would do the explaining.)

The [`clock_configure()`](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_clocks/clocks.c#L43) function can be used to set up any of the clock. In that function there's a [pair](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_clocks/clocks.c#L74) [of](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_clocks/clocks.c#L105) `hw_{clear,set}_bits()` calls that write `CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS` to the clock's `CTRL` register.

But the value of `CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS`, `0x00000800`, does not make sense with `clk_sys` for example, because in that clock's `CTRL` register the bits 31:8 are marked as "Reserved" ([2.15.7. List of Registers](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf)).

lurch, 2021-05-16:
I'm afraid I dunno the answer to your question, but in the meantime I **can** tell you that `CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS` relates to this part of the datasheet:
[image]

Ahhh.... perhaps the fact that it's explicitly using `CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS` is in fact a red-herring, and the code is simply relying on the fact that _every_ clock has its `ENABLE` bit at the same bit-position as `CLOCKS_CLK_GPOUT0_CTRL_ENABLE_BITS` :slightly_smiling_face: (which is a safe assumption to make, as all the clocks registers are generated programatically).

kilograham, 2021-05-16:
> EDIT: And writing to a non-existent part of a register probably has no effect? So it's probably easier to just "write to the clock's ENABLE field (regardless of whether it exists or not)" than it is to "check if the clock has an ENABLE field, and then write to it" ? man_shrugging

Which I think was the original question :-)

kahara, 2021-05-16:
Thank you, I'll add that as a comment next to those calls and make a PR.

kahara, 2021-05-16:
Although with the resus clock bit twelve is `FORCE`, meaning "Force a resus, for test purposes only" and bit eight is `ENABLE`. Supposedly this enforcement isn't too harmful then.

Wren6991, 2021-05-17:
Yes that code assumes that that bit has no effect on the always-on clocks `clk_sys` and `clk_ref`, where it is reserved. Not a good assumption to make in general, but it is true for this hardware.

It is good to call this out explicitly, thank you for the PR.

Wren6991, 2021-05-17:
#414 is merged

commit 5384b6b4 (2021-05-17): add comment about using clk_gpout0 enable bit (Fixes #413)
closed 2021-05-17 by Wren6991